esp-rs/esp-hal #2: Implement `TIMG`, `UART`, `GPIO` peripherals for all chips

jessebraham, 2021-11-29:
As an MVP, we should have the `TIMG`, `UART`, and `GPIO` peripherals implemented for each chip, __with working examples__. Presently this is all implemented to some degree for the ESP32-C3.

The remaining chips should be able to use the `TIMG` and  `UART` implementations with little to no modifications. `GPIO` will likely have to be device-specific.

- [ ] ESP32
    - ~`TIMG`~, ~`UART`~, `GPIO`
- [x] ESP32-C3
    - ~`TIMG`~, ~`UART`~, ~`GPIO`~
- [ ] ESP32-S2
    - `TIMG`, `UART`, `GPIO`
- [ ] ESP32-S3
    - `TIMG`, `UART`, `GPIO`

bjoernQ, 2021-12-06:
Thanks for your efforts - good to see something going on here 🙂 

I was trying a bit with this here: https://github.com/bjoernQ/esp-hal/tree/esp32-local-pac

Basically, what I was trying is to have the GPIO module living in the _esp-hal-common_ for ESP32C3 and ESP32 (don't have access to S2, S3 currently). Not sure if that will ever work but I have at least digital IO working for ESP32 and ESP32C3 (at least blinky blinks - not tested digital input yet)

I have a question btw about the evolution of the ESP32-PAC. In my fork I use the SVD from `https://github.com/espressif/svd` since that matches the other SVDs there more. However, it seems you are using the generated SVD from `https://github.com/esp-rs/esp32` - are you avoiding the "official" SVD on purpose?

bjoernQ, 2021-12-06:
Thanks for the clarification regarding the SVD / PAC. I noticed the fork but didn't check the actual SVD file.
My spare time is currently a bit limited unfortunately but I hope to get the ESP32 GPIO working

jessebraham, 2021-12-08:
There's no rush, I appreciate the effort! If there's anything I can do to help expedite the process please let me know.

bjoernQ, 2022-01-05:
I created a PR for the GPIO implementation in _esp-hal-common_: https://github.com/jessebraham/esp-hal/pull/3

jessebraham, 2022-01-10:
Regardless of our current lack of runtime support, I have begun implementing the `GPIO` peripheral for the ESP32-S2 and ESP32-S3 as well. I still need to add the various alternate functions, and verify that things build (PACs might needs some patching), but the work so far can be found in this branch:  
https://github.com/jessebraham/esp-hal/tree/feature/gpio-s2-s3

jessebraham, 2022-02-23:
The three aforementioned peripherals are now supported by all chips.

closed 2022-02-23 by jessebraham