esp-rs/esp-hal #4: Add one or more `procmacros` crates

jessebraham, 2022-01-06:
I'm specifically interested in the `#[ram]` and `#[interrupt]` attributes that are generally implemented by the `procmacros` crate. An example of this can be seen in the [esp32-hal repository](https://github.com/esp-rs/esp32-hal/tree/master/procmacros).

I imagine this may be tricky to get working for all chips, so it's possible more than one implementation may be necessary. I have not looked into this enough to know whether or not that is the case at this time.

bjoernQ, 2022-01-10:
I plan on looking into the _ram_ macro. Seems the ESP32C3 linker scripts need some changes. I'd also like to look into an easy way to support both, direct-boot and non-direct-boot.

The _interrupt_ macro probable makes more sense together with the general implementation of interrupt support

jessebraham, 2022-02-14:
Closed by #6 and #10

closed 2022-02-14 by jessebraham