esp-rs/esp-hal #19: Track peripheral driver implementations for all chips

jessebraham, 2022-02-23:
The table below enumerates all peripherals for each chip supported by `esp-hal`. Support in this case does not mean a feature-complete and bug-free driver implementation, only that you are able to interact with the peripheral in some meaningful way.

If a cell contains am **em dash** (&mdash;) this means that the particular peripheral is not present for a chip. A **check mark** (✓) means that some driver implementation exists.

The table should be complete, though some minor naming changes have been made for consistency. If there are any missing peripherals or errors please comment below indicating such.

If you would like to contribute to this project, we ask that you please comment below "claiming" a peripheral, just to help ensure people are stepping on each others' feet.

| Peripheral |  ESP32  | ESP32-C3 | ESP32-S2 | ESP32-S3 | Claimed By |
| ---------- | :-----: | :------: | :------: | :------: | :--------: |
| ~~GPIO~~   |    ✓    |    ✓     |    ✓     |    ✓     |            |
| LEDC       |         |          |          |          |            |
| ~~UART0~~  |    ✓    |    ✓     |    ✓     |    ✓     |            |
| ~~UART1~~  |    ✓    |    ✓     |    ✓     |    ✓     |            |
| ~~UART2~~  |    ✓    | &mdash;  | &mdash;  |    ✓     |            |
| USB        | &mdash; |          |          |          |            |
| I2C0       |         |          |          |          |            |
| I2C1       |         | &mdash;  |          |          |            |
| I2S0       |         | &mdash;  |          |          |            |
| I2S1       |         |          | &mdash;  |          |            |
| LCD_CAM    | &mdash; | &mdash;  | &mdash;  |          |            |
| ~~TIMG0~~  |    ✓    |    ✓     |    ✓     |    ✓     |            |
| ~~TIMG1~~  |    ✓    |    ✓     |    ✓     |    ✓     |            |
| PWM0       |         | &mdash;  | &mdash;  |          |            |
| PWM1       |         | &mdash;  | &mdash;  |          |            |
| PWM2       | &mdash; | &mdash;  | &mdash;  |          |            |
| PWM3       | &mdash; | &mdash;  | &mdash;  |          |            |
| UHCI0      |         |          |          |          |            |
| UHCI1      |         | &mdash;  |          |          |            |
| RMT        |         |          |          |          |            |
| PCNT       |         | &mdash;  |          |          |            |
| SPI        |         |          |          |          |            |
| SPI2       |         |          |          |          |            |
| SPI3       |         |          |          |          |            |
| SDMMC      |         | &mdash;  | &mdash;  |          |            |
| SDIO_SLAVE |         | &mdash;  | &mdash;  | &mdash;  |            |
| TWAI       |         |          |          |          |            |
| RNG        |         |          |          |          |            |
| EMAC       |         | &mdash;  | &mdash;  | &mdash;  |            |
| WIFI       |         |          |          |          |            |
| BT         |         |          | &mdash;  |          |            |
| WIFI_BT    |         |          |          |          |            |
| BT_BB      |         |          | &mdash;  |          |            |
| BT_LC      |         |          | &mdash;  |          |            |
| AES        |         |          |          |          |            |
| RSA        |         |          |          |          |            |
| SHA        |         |          |          |          |            |
| HMAC       | &mdash; |          | &mdash;  |          |            |
| DS         | &mdash; |          | &mdash;  |          |            |
| SYSTIMER   | &mdash; |          |          |          |            |
| DMA        |         |          |          |          |            |
| DEDIC_GPIO | &mdash; | &mdash;  |          |          |            |
| SARADC     |         |          |          |          |            |

bjoernQ, 2022-02-25:
working on I2C

jessebraham, 2022-03-04:
I have an `RNG` implementation complete which I will PR soon, waiting on PACs to be updated.

bjoernQ, 2022-03-07:
I will start implementing `embedded_hal::spi::FullDuplex`

fkohlgrueber, 2022-04-23:
@ducktec I just saw that you're working on supporting the `rmt` peripheral too right after I opened https://github.com/esp-rs/esp-hal/issues/52. Are you still working on the topic? If yes, we could probably join forces.

ducktec, 2022-04-25:
@fkohlgrueber I'm already pretty far with the implementation of the `OutputChannel` functionality (I'm not planning to support the input functionality as part of my PR so far).

For better visibility I have just created #53. The PR so for includes the `ESP32-C3` and `ESP32-S2` variants. The `ESP32` and `ESP32-S3` variants are still WIP. The `ESP32-S3` variant is missing the RMT peripheral in the SVD altogether right now.

The challenge is that all variants have a sightly different RMT feature set and register structure. This makes bringing them all together into generic implementation a lot of fun. 😆

gustavonihei, 2022-08-22:
RTC Watchdog Timer has been implemented and validated on https://github.com/esp-rs/esp-hal/pull/134 and https://github.com/esp-rs/esp-hal/pull/139 for all currently supported chips.

The RTC driver itself has been improved as part of those tasks, but some pieces are not yet implemented, e.g. configuration for sleep modes.

alexbohm, 2022-09-07:
Working on the TWAI and the associated `embedded_hal::can::Can` traits.

liebman, 2022-09-23:
This is missing MCPWM?

jessebraham, 2022-09-23:
@liebman I think that's what those were supposed to be, I honestly don't even remember where I generated this list from at this point 😁 Thanks for pointing that out. I've updated the table to use `MCPWM` instead though, as based on the cells which were blocked out it seems to fit.

jessebraham, 2022-10-20:
`ADC` is supported on all chips now that #217 is merged.

bjoernQ, 2022-10-24:
Going to implement USB (device) for ESP32-S3 and S2 - starting with S3

jessebraham, 2022-10-28:
We now have some level of `DMA` support for all supported chips.

bjoernQ, 2022-11-03:
Going to implement first steps to get I2S support (starting with ESP32-C3 / ESP32-S3 - then ESP32 and ESP32-S2, just TX for the start and only supporting standard I2S - we can improve that later but we need something to start with)

dimpolo, 2022-11-04:
Working on `MCPWM`

dimpolo, 2022-11-07:
Do I understand correctly that the ESP32-C2 is secretly an ESP8684? (what's up with espressif's chip names? 😅 )

If so, the first table seems to be missing some `—` for the ESP32-C2.

i404788, 2022-11-08:
Looking into implementing `SHA`

jessebraham, 2022-11-09:
> Do I understand correctly that the ESP32-C2 is secretly an ESP8684? (what's up with espressif's chip names? 😅 )
> 
> If so, the first table seems to be missing some `—` for the ESP32-C2.

My understanding regarding the device names is that the ESP8684/ESP8685 are variants of the ESP32-C2/C3 respectively with flash memory integrated into the package. I'm not aware of any other differences, though that's not to say they don't exist.

You are correct though, seems I updated one table's column but not the others' when I added the C2. I will get that fixed shortly!

raiker, 2022-11-12:
I'd like to tackle USB serial and I2S on the ESP32-C3. I'm going to start by fleshing out the USB_SERIAL_JTAG so it looks and feels a bit more like a UART (Rx, interrupts, etc).
I've also got ESP32 hardware, so I might be able to get some I2S testing done on that, but -C3 is my primary platform.

jessebraham, 2022-11-17:
Preliminary `I2S` support added in #262

jessebraham, 2022-11-28:
`SHA` support added in #257

bjoernQ, 2022-12-01:
`MCPWM` support added in #255

liebman, 2022-12-31:
I'm looking at PCNT.

akashlama1998-icloud, 2023-01-06:
No AES, RSA and I guess secure boot as well that means it's not production ready yet 😔. It's really early I guess. Just suited for DIY right now rather than shipping to production. I'm eagerly for the project to be stable and when I search for esp rs or esp for rust the official document comes up rather than something else. 😌

akashlama1998-icloud, 2023-01-06:
So, can I do secure boot and encryption, flash encryption with esp-idf? And when will be the bare metal be stable?

akashlama1998-icloud, 2023-01-06:
And The Rust on ESP Book is fine for the starting point.. are there any extensive documentations available?

akashlama1998-icloud, 2023-01-08:
Lack of documentation I get it. But SECURE BOOT for esp32 in rust how can this be off topic. It's an essential part if that doesn't exist or hasn't been a plan to support then you can only use rust on esp32 for home DIYs NOT production.

akashlama1998-icloud, 2023-01-08:
I tried to research for it myself couldn't get it. You could've atleast told me there are plans or something or anything rather than marking it as an off topic comment.

akashlama1998-icloud, 2023-01-08:
I you knew anything about on this topic you could've atleast nudge me towards the right direction. Helping me make the deicision if now the time is it suitable for production or not.

akashlama1998-icloud, 2023-01-08:
If you knew anything about on this topic you could've atleast nudge me towards the right direction. Helping me make the deicision if now the time is it suitable for production or not.

akashlama1998-icloud, 2023-01-08:
If you knew anything about on this topic you could've atleast nudge me towards the right direction. Helping me make the deicision if now is the time... or is it suitable for production or not.

MabezDev, 2023-01-08:
They were marked as off-topic because they are off-topic for a tracking issue thread. Many people are subscribed to this issue and don't want to be spammed with off-topic questions.

Please open a discussion thread/issue or join the matrix channel to ask questions like these, we are more than happy to answer them :).

akashlama1998-icloud, 2023-01-08:
Ok.. I'll do so..

matrixhead, 2023-01-10:
i would like to work on AES in esp32

i404788, 2023-01-16:
Does anyone from espressif have documentation on `LCD_CAM`? I would like to work on implementing it but I can't find much of how it works outside of the pacs/registers.

liebman, 2023-01-16:
Download the [technical reference manual ](https://www.espressif.com/en/support/documents/technical-documents?keys=&field_download_document_type_tid%5B%5D=963)for the chip in question, it will have a section on each peripheral.

jessebraham, 2023-01-17:
`PCNT` added in #328

liebman, 2023-01-20:
I think that HALL_SENS peripheral should be remove from the list.  Justification: [PCN20221202](https://www.espressif.com/sites/default/files/pcn_downloads/PCN20221202%20Remove%20Hall%20Sensor%20from%20ESP32%20Series%20of%20Documentation.pdf)

MabezDev, 2023-01-20:
I would agree, we also removed it from esp-idf-hal (from v5 onwards).

jessebraham, 2023-01-24:
`AES` support added in #348

jessebraham, 2023-01-25:
I've regenerated the tables in the first comment from the current PACs, and filled in each cell by hand.

There are a number of new peripherals in the first table now, however I'm not sure if all of them actually require peripheral drivers. I trimmed this down already from the generated version, but will need to spend some time in the TRMs to figure out which other rows can be eliminated, if any.

If you see anything in the first table that does not belong there please let me know so I can remove it.

matrixhead, 2023-01-27:
working on RSA in esp32

jessebraham, 2023-02-27:
ESP32-C6 support was added in #392, and the tables have been updated. However, there are a number of new peripherals which have not yet been added; I'll do this at some point.

nabilmerk, 2023-02-28:
@matrixhead Are you still actively working on RSA driver ? I could give some help if you want.
I'm also willing to work on ECC and HMAC driver but I would need some time for testing as I need to order some boards.

matrixhead, 2023-02-28:
@nabilmerk yes i'm still working on it and i'm more than happy to discuss this with you.

jessebraham, 2023-02-28:
@nabilmerk that would be great, thank you for offering! There's no rush at all, grateful for any contributions 😁

nabilmerk, 2023-03-02:
@jessebraham the basic function for computing HMAC are ready for testing but in order to use the HMAC, efuse must be programmed, even for upstream mode. Looking at the efuse module, I think I will also write the HAL for this module except if someone is on it?

jessebraham, 2023-03-02:
@nabilmerk thanks for the offer, that'd be great. I don't believe anybody is working on anything `EFUSE` related, all we really have in the repo is a function to read the MAC address, so go nuts.

bjoernQ, 2023-03-17:
I'm looking into half-duplex SPI including dual and quad SPI

nabilmerk, 2023-04-02:
After digging around with EFUSE, I came to the conclusion that the EFUSE is a peripheral that we should set only in 'factory' environment. If someone's look at ESP-IDF, they actually don't ship software for writing EFUSE but they ship software for programming them when the board is attached to the PC.
Except if someone really needs EFUSE driver, I would finally propose that the driver should not be implemented, at least, for writing datas. What do you think ?

MabezDev, 2023-04-03:
> I would finally propose that the driver should not be implemented

I agree, we shouldn't have write access to EFUSE on chip. We already have _some_ read-only access in the soc/$chip/efuse.rs modules, whether this is enough to mark them it as complete or not is a different question.

AnthonyGrondin, 2023-08-09:
I was looking for it 5 minutes ago to see what still needed to be implemented and what has been implemented. Haha