esp-rs/esp-hal #21: GPIO `InputSignal` and `OutputSignal` enums are not convenient to use

jessebraham, 2022-02-24:
This needs to be addressed sooner than later, as it will only cause more pain as we continue to add peripheral drivers. Context is available in this issue: https://github.com/esp-rs/esp-hal/issues/5

One possibility is added the enumerated values directly to the SVDs, such that they would be included in the PACs.

bjoernQ, 2022-02-25:
Having those definitions in the SVD sounds like a good idea. Might be also useful outside the Rust community.
Would be a good mid-term / long-term solutions since it also requires support in the used tools.

Short term I will just move those enums one level up in the dependency hierarchy (into common-hal). Once we get those enums from the PAC it's not a big change this way

jessebraham, 2022-03-02:
Closed by #23