esp-rs/esp-hal #29: Document all packages and peripherals

jessebraham, 2022-03-07:
It's that job nobody wants to do!

This is a long-term project, as there's not much point documenting drivers until their interfaces are relatively stable.

Each HAL package should have top-level documentation at the very least. Peripheral drivers should additionally be documented, preferably even containing simple examples as doc tests.

IshanDaga, 2022-11-16:
Hi Jesse, 
New to Rust, though I have been using the ESP-IDF for the last 15 or so months.

I'd like to be able to contribute to this project and learn rust while I'm at it - is this a good place to start?

jessebraham, 2022-11-16:
@IshanDaga if you are willing to work on this then yes, that would be awesome! This is definitely one of our weaker points currently so it'd be great to improve this.

In general (with the exception of the `chip support` label) any open issue is fair game; if it's assigned to somebody already just ask if they're actively working on it, there's a good chance they are not and that you can take it over 😁

jessebraham, 2023-07-12:
I've documented the `Rng` peripheral in #650, which can be loosely used as a model for other peripherals. @playfulFence is also working on a number of drivers' documentation.

jessebraham, 2023-08-09:
With #680 merged I'm happy to call this "done". There is of course more work to do, but all packages and modules have at least some documentation now.

closed 2023-08-09 by jessebraham