esp-rs/esp-hal #44: We need a way to configure clock rates

bjoernQ, 2022-04-05:
In the first iteration it would be sufficient to just configure all clocks and freeze them

bjoernQ, 2022-04-27:
The simplest to start with would be a "boot-default" configuration without actually changing any clock settings. This way drivers can get the clock rates and as soon as we actually make it possible to change clock rates things should "just work"

bjoernQ, 2022-05-10:
I wonder which clocks need to be contained in the clock config struct?

I can think of these:
- CPU CLK
- APB_CLK
- Option: XTAL_CLK
- Option: FOSC_CLK
- Option: RTC8M
- Option: APLL_CLOCK
- Option: REF_TICK
- Option: PLL_160M
- Option: PLL_240M
- Option: PLL_CLK
- Option: PLL_D2_CLK

Probably there are more.

The "Option" prefixed ones are not available for all targets. Should they be wrapped by an _Option_ in the struct or just not be contained if not available for the target?

MabezDev, 2022-05-10:
> Option: XTAL_CLK

Is this really an option? I thought all esp's had some form of crystal clock? It's also the default clock source for a lot of peripherals (at least on the c3, when I skimmed the TRM).

I haven't looked through all the chip trms but this looks like a solid base (maybe we can do some smart grepping of the SVDs to find all clock sources?). It should be easy to determine what clocks are available to the driver and pick the best one. It might get interesting for peripherals like LEDC, where the input clock puts a limitation of certain output frequencies... but we'll worry about that later :laughing:.

bjoernQ, 2022-05-10:
>> Option: XTAL_CLK

> Is this really an option? I thought all esp's had some form of crystal clock? It's also the default clock source for a lot of peripherals (at least on the c3, when I skimmed the TRM).

Ah true - I just looked at the "Peripheral Clock Usage" table and e.g. S2 doesn't list it as a direct clock source for peripherals _but_  it's an indirect clock source ... 🕵️‍♂️ good point - we should include it

bjoernQ, 2022-06-09:
First part (basically introducing `ClockConfig` with boot-defaults) has been done in #77 

Next task is adding support for actually changing the clock rates

bjoernQ, 2022-07-21:
CPU clock settings for ESP32 and ESP32-C3 are implemented in #110 

Will continue with ESP32-S2 and ESP32-S3 then ... after that I would consider this to be done

PR #111 (2022-07-22): CPU clock settings for ESP32-S2 and ESP32-S3
This adds a way to configure the CPU clock for ESP32-S2 and ESP32-S3 to 80,160 and 240 MHz.

This is a non-breaking change.
I didn't add any examples for this since they would be pretty boring

Fixes #44
closed 2022-07-22 by jessebraham