esp-rs/esp-hal #46: Multicore support for ESP32 and ESP32S3

bjoernQ, 2022-04-05:
We should have ways to start and run code on the second core on dual-core chips

jessebraham, 2022-04-28:
Duplicate of #28

bjoernQ, 2022-07-04:
This is not a duplicate of the mentioned issue - starting to work in this

bjoernQ, 2022-07-04:
@jessebraham @MabezDev I started to work on this and was wondering if I should spawn the code on app-cpu in the same way ESP32-HAL did, or if I should try to use a closure .... seems to work but unsure if I like it so here is how the new example looks like now
```rust
#![no_std]
#![no_main]

use core::sync::atomic::{AtomicI32, Ordering};

use esp32_hal::{clock::ClockControl, pac::{Peripherals, TIMG1}, prelude::*, CpuControl, RtcCntl, Timer};
use esp_println::println;
use nb::block;
use panic_halt as _;
use xtensa_lx::mutex::Mutex;
use xtensa_lx_rt::entry;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take().unwrap();
    let system = peripherals.DPORT.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let mut timer0 = Timer::new(peripherals.TIMG0, clocks.apb_clock);
    let mut timer1 = Timer::new(peripherals.TIMG1, clocks.apb_clock);
    let mut rtc_cntl = RtcCntl::new(peripherals.RTC_CNTL);

    // Disable MWDT and RWDT (Watchdog) flash boot protection
    timer0.disable();
    timer1.disable();
    rtc_cntl.set_wdt_global_enable(false);

    timer0.start(1u64.secs());
    timer1.start(500u64.millis());

    let counter = xtensa_lx::mutex::SpinLockMutex::new(AtomicI32::new(0));

    let mut cpu_control = CpuControl {};
    let mut cpu1_fnctn = || { cpu1_start(&mut timer1, &counter); };
    let _guard = cpu_control.start_app_core(&mut cpu1_fnctn).unwrap();    

    loop {
        block!(timer0.wait()).unwrap();

        let count = (&counter).lock(|counter| counter.load(Ordering::Relaxed));
        println!("Hello World - Core 0! Counter is {}", count);
    }
}

fn cpu1_start(timer: &mut Timer<TIMG1>, counter: &xtensa_lx::mutex::SpinLockMutex<AtomicI32>) -> ! {
    println!("Hello World - Core 1!");
    loop {
        block!(timer.wait()).unwrap();

        (&*counter).lock(|counter| {
            counter.store(
                counter.load(Ordering::Relaxed).wrapping_add(1),
                Ordering::Relaxed,
            );    
        });
    }
}

```

I like that there are no `static mut`s  but
```rust
    let mut cpu1_fnctn = || { cpu1_start(&mut timer1, &counter); };
    let _guard = cpu_control.start_app_core(&mut cpu1_fnctn).unwrap();
```

Doesn't look too nice. The need to declare that `cpu1_fnctn` (to keep it alive when passing the reference to `start_app_core`) is probably understandable and not that weird.

But not so nice is that I can't move the `timer1` into the closure because I struggle to call the `FnOnce` via the reference.

It's probably not that bad but before going further into that direction I just wanted to hear your opinions on that - if you don't like that I can revert that. On the other hand we can also keep that and see if it works good enough and change back if not ... it's not that much work

PR #96 (2022-07-05): Implement starting a task on second core of ESP32 and ESP32-S3
This implements a way to start a task on the app CPU of ESP32 and ESP32-S3.

On ESP32-S3 this currently needs to use a ROM function since the register for the CORE1 boot address can't be found in TRM or SVD. Not a big deal but would be nice to get rid of the ROM function one day.

In this PR I tried to use closures for the APP CPU task - let us see how good that works ... if it doesn't for some reason we can revert to having a free-standing function that gets started and needs to get it's resources via statics.

Fixes #46
closed 2022-07-05 by jessebraham