esp-rs/esp-hal #54: GPIO interrupts funky (tested on ESP32)

orangecms, 2022-04-25:
### Example

The code in https://github.com/esp-rs/esp-hal/blob/43c8f34e5f532bf6edd5cc23fef8e27c7a60c571/esp32-hal/examples/gpio_interrupt.rs describes the following setup:

- configure GPIO0 (button) as input with internal pull-up, listening on falling edge
- enable GPIO interrupt 1, level-triggered
- enable GPIO interrupt 1 in mask
- define `level1_interrupt` function which clears the interrupt and prints to serial

This raises the following questions already:
- Why configure the button to listen on falling edge, but use the level-triggered interrupt 1?
- Why not use an edge-triggered interrupt then, such as interrupt 10?

I have taken the example code and tried to adapt it. However, I couldn't get it to work as I intended.

### Scenario

We have the following board based on the ESP32 D1 mini module:
https://github.com/das-labor/badge-2021

This board has 4 buttons with debounce circuits each (see also the schematics, linked in README). Two are push buttons and two more are part of joysticks. They are routed to GPIOs 0, 2, 5, and 12.

Note: Those buttons work just fine with MicroPython, checking their level in a loop.

### Code

I have iterated over multiple attempts, starting from the example. See https://github.com/das-labor/badge-2021-rs/commit/e13aa27567aed2d548d5f68e0e2d07503a0aaefb and also the history before that commit.

What I expected to work:
- set up an interrupt to be triggered on any edge, setting internal pull-up
- enable interrupt 10 (edge-triggered) for GPIO on ProCpu
- enable interrupt 10 in mask
- define `level10_interrupt`

So far I can never see the `level10_interrupt` executed. However, even if not enabled, I _do_ see the `level1_interrupt` executed, and it only works for one button, IO0. If listening to low level instead of listening on edge, it is constantly triggered without pressing the button.

This is all quite confusing. Any ideas would be highly appreciated. :pray:

orangecms, 2022-04-26:
Looking through the linker scripts and various repos - hal, pac, rt -, I am a bit confused that only interrupt functions 1-7 are defined.
So I need to extend the linker script? And how would I do that?

https://github.com/esp-rs/xtensa-lx-rt/blob/master/exception-esp32.x.jinja Is this what I need to do, map to the low-level functions?

orangecms, 2022-04-26:
Browsing other issues, I also found this, is it related?

https://github.com/espressif/svd/issues/6

bjoernQ, 2022-04-26:
So, I gave your code a try and it seems like there is an issue with GPIO5 for me but the others are working.
I modified your code here: https://gist.github.com/bjoernQ/71999fd43eba4680a09a9870eb7e9311
Maybe you can give that a try and see if the inputs beside GPIO5 is working for you

bjoernQ, 2022-04-26:
Looking further into this it seems it's more an issue of configuring pull-downs / pull-ups than an issue with interrupts

orangecms, 2022-04-26:
Thank you a lot - I will adhust the title and try your suggestions. :-)
Since I want to check for edge triggers, I understand now that I need to use a lower priority interrupt handler for that, because level 1 is not available for that. The relationship between the handler numbering and the level was indeed not clear to me. Thank you so much!

I'll also see that I file some PRs against the hal repo to simplify and explain the example. :-)

bjoernQ, 2022-04-27:
@orangecms would you mind posting your working MicroPython code? Thanks

orangecms, 2022-04-27:
> @orangecms would you mind posting your working MicroPython code? Thanks

Sure :)

https://github.com/das-labor/badge-2021/blob/main/sw-demo/spi_rects.py#L12

bjoernQ, 2022-04-28:
There is definitely one thing that looks wrong: https://github.com/esp-rs/esp-hal/blob/eb47b0fe212f0cf7ce1a6564478058b2f48c2bcb/esp-hal-common/src/gpio.rs#L506

PD should be `true` here

Still that doesn't solve the problem for all the pins as it seems

bjoernQ, 2022-04-28:
It seems like #55 should fix this

orangecms, 2022-04-28:
With the changes in https://github.com/orangecms/esp-hal/tree/gpio-input-pullup-fix the two GPIOs 0 and 5 are working now.
:partying_face: 

So I guess that 2 and 12 are special as mentioned in #55.

bjoernQ, 2022-05-02:
Good to hear that - GPIO 2 and 12 issues should be solved by #55 - hope to get to that soon

Also, thanks for raising this issue - it's very valuable

bjoernQ, 2022-05-02:
@orangecms I have put together something here: https://github.com/bjoernQ/esp-hal/tree/bugfix/errata_3_6
Would be great if you can find some time to give this a try - hopefully this will fix the issue

orangecms, 2022-05-02:
> @orangecms I have put together something here: https://github.com/bjoernQ/esp-hal/tree/bugfix/errata_3_6 Would be great if you can find some time to give this a try - hopefully this will fix the issue

YES! I am happy to report that indeed this fixes both GPIOs 2 and 12 :partying_face: Thank you so much! \o/

orangecms, 2022-05-02:
Just one note: They also both trigger once initially. That can be worked around, but might be surprising.

bjoernQ, 2022-05-03:
Greate - thanks for giving it a try. Will create a PR for this

closed 2022-05-03 by MabezDev