esp-rs/esp-hal #58: Reduce Macro Usage for GPIO Implementation

bjoernQ, 2022-05-02:
Currently we use macros nested multiple levels for implementing GPIO.

We should reduce that since it's not convenient to work with (on the implementation side).

Ideally this
- shouldn't change anything on the call-site
- reduce the macro usage to just one top level macro (or as few levels of nesting as possible)

Note: We shouldn't touch this before #55 since that one will introduce one more macro to get rid of

bjoernQ, 2022-07-22:
@jessebraham do we have more ideas what to do besides what is done in #78 ? I personally don't have any - so this can be closed?

jessebraham, 2022-07-22:
Yeah I'm with you, I've spent some time banging my head against this problem with no real results.

closed 2022-07-22 by jessebraham