esp-rs/esp-hal #65: Port over existing peripheral drivers from `esp32-hal`

jessebraham, 2022-05-27:
There are a number of peripheral drivers already implemented in the [esp32-hal](https://github.com/esp-rs/esp-hal) repository. Since this crate has been published, we aren't really able to release a new version until we've roughly reached feature-parity with the previous implementation. Because of this, we should begin working on porting this drivers.

These can either just be dumped in the `esp32-hal` package for now, making the required changes just to get things building, or can be re-written to be compatible across all supported chips. While the latter method is preferable, the prior is likely the more realistic path forward.

jessebraham, 2022-05-30:
The existing drivers are:

- [ ] `analog`
- [ ] `alloc`
- [ ] `clock_control`
- [ ] `delay`
- [ ] `dport`
- [ ] `dprint`
- [ ] `efuse`
- [ ] `external_ram`
- [ ] `gpio`
- [ ] `i2c`
- [ ] `interrupt`
- [ ] `ledc`
- [ ] `mem`
- [ ] `serial`
- [ ] `spi`
- [ ] `timer`

bjoernQ, 2022-06-01:
I think `dprint` shouldn't get ported since we have `esp-println` for that.

For clock control we also already have issue #44 

I'm not sure if this really should be a hard requirement for a potential release. Someone using the deprecated HAL can just pin the version and even more important even if we have all of those the new HAL isn't a drop-in replacement since many things are just different (for good reasons).

But certainly, it would be a good thing to cover all of those

jessebraham, 2022-07-20:
I'm going to close this. Other than `LEDC` (which is in progress and hopefully will be merged in the coming weeks) there's not really anything left worth worrying about right now, I don't think.