esp-rs/esp-hal #69: Implement the `embedded-hal` alpha version Traits

jessebraham, 2022-06-02:
It should be possible for these to coexist with the existing `0.2.7` implementations. I have already begun work on this.

jessebraham, 2022-06-14:
A little over half of these traits have been implemented in #82.

jessebraham, 2022-07-04:
A couple more SPI traits implemented in #95.

jessebraham, 2022-08-19:
Additional SPI traits implemented in #101.

har7an, 2022-08-22:
Just a heads up: According to the maintainer of `embedded-hal-compat`, a new release may be due soon that will add compat for `1.0.0-alpha.8 <-> 0.2.7`. Maybe the old traits can be dropped in favor of using the compatibility traits then?

jessebraham, 2022-08-22:
Thanks for the update, I am planning on doing that once we get a stable release of `embedded-hal`. I believe `embedded-hal` is still going to have a `embedded-hal-nb` package split out, so I'd like to just wait until things are stable.

jessebraham, 2022-08-30:
`SpiDevice` has been implemented in #106

MabezDev, 2023-05-04:
> I think SpiDevice holding a critical section throughout a complete transaction is extremely inefficient.

I agree, this also means we can't currently reuse the device/bus struct's for async versions because it's possible to hold `RefCellMut` borrows across await points and you'll get a panic if another bus device tries to borrow. We should change this to a proper Mutex to avoid these issues.

bugadani, 2023-05-04:
> We should change this to a proper Mutex to avoid these issues.

I suspect this will be pretty difficult. I'm not actually sure about details, but if it's possible to use two SpiDevice instances on two RTIC tasks (1 for each), it will be trivially easy to deadlock a firmware - one task waiting for the other to release the Spi bus, without preemption that would make it actually possible. A critical section at least prevents this, though again, I'm not sure if the use case is actually possible or not.

jessebraham, 2023-08-22:
The `embedded-hal-*` ecosystem has moved on to `1.0.0-rc.1`, which we already include and implement, so I am going to consider this issue completed and close it.