esp-rs/esp-hal #72: Timer should use a Duration not a plain U64 as timeout

bjoernQ, 2022-06-08:
We should use a Duration instead of a plain U64 for the timer timeout.
Also we should pass in the clock config from #44 to the constructor and have a way to configure the rate of the timer ticks.

jessebraham, 2022-06-08:
Just as a note there is also the `fugit_timer` package which has its own [`Timer`](https://docs.rs/fugit-timer/latest/fugit_timer/trait.Timer.html) trait which can be implemented. Not sure how useful this is in practice, however.

jessebraham, 2022-06-28:
Is there actually a way to configure the timer tick rate other than simply selecting the clock and divider values? The only fields I'm really seeing for configuration in this regard are:

- `TIMG_T0_USE_XTAL` - 1: Use XTAL_CLK as the source clock of timer group. 0: Use APB_CLK as the source clock of timer group. (R/W) [Default: 0]
- `TIMG_T0_DIVIDER` - Timer 0 clock (T0_clk) prescaler value. (R/W) [Default: 1]

Not seeing any other relevant inputs in the block diagram either:

[image]

If these are the fields that you were referring to please let me know, I'm sure I'm missing something here :grin:

bjoernQ, 2022-06-29:
Yes, I think that is all that can be configured about the clock rate. 

My idea was just to not use `type Time = u64;` but have `Time` a fugit duration type. Since in user code it's not clear what a tick is actually (and can be different in different configurations).  So, we probably also need to take a reference to the frozen clocks in the `new` constructor. 

There is also `load_alarm_value` but I think that is okay to be timer-ticks. Maybe we should add a comment there about that, then.

PR #91 (2022-06-30): Convert the `Timer` to take time units for countdown instead of `u64`
Closes #72.
closed 2022-06-30 by bjoernQ