esp-rs/esp-hal #83: RMT driver no longer works for some chips after updating PACs to use `svd2rust` 0.24

jessebraham, 2022-06-14:
For whatever reason this update seems to have broken this driver for some of the chips. I have verified it still works on the ESP32-S2, but it has stopped working on the ESP32-C3 and ESP32-S3 for me. I do not have an ESP32 board with a neopixel on it so I cannot test this.

This obviously needs to be fixed.

jessebraham, 2022-06-14:
@bjoernQ whenever you get a chance would you be able to test the `esp32-hal/examples/hello_rgb.rs` example please?

bjoernQ, 2022-06-14:
checked on ESP32 ... it lights up three random RGBs in colors that totally don't make sense .... so, I'd say it's broken

bjoernQ, 2022-06-15:
After some investigation it pretty much looks like a bug in `svd2rust` to me.

https://github.com/rust-embedded/svd2rust/issues/616

bjoernQ, 2022-06-15:
OK - seems that - `w.ch_tx_thr_event_int_clr::<7>().set_bit() ` is not the same as `w.ch_tx_thr_event_int_clr(7).set_bit()`
I'll look into taking this into account and fix the issue on our side

PR #84 (2022-06-15): Fix RMT driver
This fixes #83
closed 2022-06-15 by jessebraham