esp-rs/esp-hal #105: `smartled` feature shouldn't be activated by default

bjoernQ, 2022-07-15:
Currently the feature is always activated as soon as a target is activated (which is always the case).

This feature must be controllable by a feature of the chip-specific HALs and not activated by default.

The example should not be build when the feature is not activated. That should be doable by something like this

```toml
[[example]]
name = "hello_rgb"
required-features = ["smartled"]
```

jessebraham, 2022-07-15:
I agree, however this unfortunately will complicate building the examples in CI.

MabezDev, 2022-07-15:
Hmm this will also need to be solved before #103 can be merged, as default `vectored` features will need to be added to the chip hals, not the common one.

PR #109 (2022-07-20): Disable `smartled` feature by default and update `xtensa-lx-rt` dependency
- Updated `xtensa-lx-rt` to the newest version (`0.13.0`)
- Disable the `smartled` feature in `esp-hal-common` and instead allow chip-specific HALs to optionally enable this
- Update the CI config to use `--features=smartled` so that the `hello_rgb` example still builds

Closes #105
closed 2022-07-20 by bjoernQ