esp-rs/esp-hal #108: Remove the usage of floats

bjoernQ, 2022-07-20:
Currently we use floats in one place: https://github.com/esp-rs/esp-hal/blob/907d43ed6f2bfde73763f9465b3e044c7777fee6/esp-hal-common/src/timer.rs#L226-L229

If possible we should remove that since it adds 2k of code (optimized)  for ESP32-C3:
```text
File  .text   Size                       Crate Name
0.0%  12.3% 1.2KiB           compiler_builtins compiler_builtins::float::div::__divdf3
0.0%   3.0%   292B           compiler_builtins __floatundidf
0.0%   1.6%   156B           compiler_builtins __floatunsidf
0.0%   1.4%   140B           compiler_builtins __gedf2
0.0%   1.4%   140B           compiler_builtins __gtdf2
0.0%   1.3%   130B           compiler_builtins __fixunsdfdi
```

bjoernQ, 2022-07-26:
note: the upcoming LEDC implementation is currently also using floats

jessebraham, 2022-07-26:
I did say to @JurajSadel that we can not worry about the floats right now, just for the sake of getting things completed and merged. If you have a problem with this please feel free to say so though, doesn't really make a difference to me.

bjoernQ, 2022-07-26:
No problem with that - we can replace the floats in one go later

PR #173 (2022-08-30): Replace all float calculations with fixed ones
closes #108
closed 2022-08-30 by bjoernQ