esp-rs/esp-hal #113: Cannot infer type of optional SPI pins

jessebraham, 2022-07-22:
I apparently didn't test this before merging (oops), but if you attempt to initialize the `SPI` peripheral without providing all pins it is impossible to build. For example:

```rust
let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
let sck = io.pins.gpio7;
let mosi = io.pins.gpio6;

let spi = Spi::new(
    peripherals.SPI2,
    sck,
    mosi,
    None,
    None,
    16u32.MHz(),
    SpiMode::Mode0,
    &mut system.peripheral_clock_control,
    &clocks,
);
```

Results in:

```text
error[E0282]: type annotations needed
  --> src/main.rs:41:9
   |
41 |         None,
   |         ^^^^ cannot infer type of the type parameter `T` declared on the enum `Option`
   |
help: consider specifying the generic argument
   |
41 |         None::<T>,
   |             +++++

For more information about this error, try `rustc --explain E0282`.
```

bjoernQ, 2022-07-23:
That is what I originally introduced the no-pin type for .

We need a similar solution here

bjoernQ, 2022-07-25:
Something like we have for serial: https://github.com/esp-rs/esp-hal/blob/34b1e09662727c0619fe787dc19ddcf08434e0a1/esp-hal-common/src/serial.rs#L134-L151

bjoernQ, 2022-07-26:
@jessebraham to solve this like we did for UART we need a bunch of constructors which take combinations of the optional pins 

while I guess for UART it was sane to have one with all four pins and one with only TX and RX I don't know how to handle this here

IMHO all 4 pins should be optional which gives 16 combinations (well 15 since "no pin at all" doesn't make sense)

maybe not all combinations are really useful but still - should I just make a suggestion of constructors and if we find something is missing, we just add it as a non-breaking change?

jessebraham, 2022-07-26:
I think that's probably the least painful path forward for now. We can always change it later if need be, not the end of the world.

jrmoulton, 2022-07-26:
As a note it is possible to construct the spi block as is. A valid gpio type needs to be given. E.g.

```rust
    let spi = hal::spi::Spi::new::<
        _,
        _,
        hal::gpio::Gpio0<Input<PullDown>>,
        hal::gpio::Gpio0<Input<PullDown>>,
    >(
        peripherals.SPI2,
        sclk,
        mosi,
        None,
        None,
        2u32.MHz(),
        hal::spi::SpiMode::Mode1,
        &mut pcc,
        &clocks,
    );

```

Maybe providing a valid No Pin type would be an easier fix. But then the user would have to provide that No Pin type in the generics

closed 2022-07-27 by jessebraham