esp-rs/esp-hal #115: Generic Pin Type

jrmoulton, 2022-07-25:
The Hal currently lacks a generic pin type. Other HAL's such as the nrf-hal use a degrade method to obtain a generic pin. While working on the SpiDevice implementation the lack of a generic pin makes it so that I can't have a method take a type of pin2 and then take a type of pin3 for example. Is there a reason that a generic pin doesn't already exist?

bjoernQ, 2022-07-25:
There no specific reason other than "nobody did it, yet"

jrmoulton, 2022-07-25:
Ok. I'll work on implementing it

jessebraham, 2022-07-25:
I was thinking about this last week briefly (didn't get around to starting on it) but at the time I wasn't sure how to handle the banked GPIO access. Different GPIO require different registers to be read/written depending on which bank they're on, so we may need to do some runtime checks to accomplish this, unless you can come up with something clever 😁

jessebraham, 2022-09-02:
Just a heads up, I'm pretty deep into a fairly extensive rework of the GPIO module, and I have a generic pin type in this work. Hoping to PR something next week.

jrmoulton, 2022-09-02:
Good to know. I tried several different clever things and none of them worked so this is good to hear

closed 2023-02-08 by bjoernQ