esp-rs/esp-hal #116: Revise APIs ported from ESP32-HAL

bjoernQ, 2022-07-25:
Some of the APIs that just get straight ported from ESP32-HAL have a potential of being used wrong.

e.g. one could try to use ADC without ever calling `enable_pin` or for the (not yet merged) LEDC one could call `getChannel` with different pins but the same channel

In many places there are runtime checks but we can try to make use of the type-system to turn them into compile-time errors where possible

jessebraham, 2022-07-25:
Would it be possible to create a task list in this issue to track which APIs need revising?

bjoernQ, 2022-07-26:
Sure - I will add the things I identified - might not be an exhaustive list however

bjoernQ, 2022-07-26:
What should get changed in a way that runtime-errors are turned into compile-time errors (or make it error out at all)

- [ ] ADC read currently does a runtime check if the given PIN is configured for ADC and panics if not
- [ ] it seems that one could assign pins multiple times to an RMT channel - there is a runtime error specified but it seems it's not checked
- [ ] LEDC (not merged yet): it's possible to create multiple instances since the constructor doesn't take the LEDC peripheral
- [ ] LEDC (not merged yet): it's possible to call `get_channel` multiple times with the same channel and different PINs

JurajSadel, 2022-09-06:
LEDC: it's possible to create multiple instances since the constructor doesn't take the LEDC peripheral: closed with #181

closed 2022-12-21 by jessebraham