esp-rs/esp-hal #120: SmartLedAdapter.write() hangs when using more than one LED

fkohlgrueber, 2022-07-26:
I tried to use an led strip with my esp32c3, but it seems like there's a hang when using more than one led.

### Minimal reproducible example

Take the `hello_rgb` example and modify it to drive two leds. Here's the diff: 

```diff
diff --git a/esp32c3-hal/examples/hello_rgb.rs b/esp32c3-hal/examples/hello_rgb.rs
index a4a3b41..2ae38d7 100644
--- a/esp32c3-hal/examples/hello_rgb.rs
+++ b/esp32c3-hal/examples/hello_rgb.rs
@@ -62,7 +62,7 @@ fn main() -> ! {
 
     // We use one of the RMT channels to instantiate a `SmartLedsAdapter` which can
     // be used directly with all `smart_led` implementations
-    let mut led = <smartLedAdapter!(1)>::new(pulse.channel0, io.pins.gpio8);
+    let mut led = <smartLedAdapter!(2)>::new(pulse.channel0, io.pins.gpio8);
 
     // Initialize the Delay peripheral, and use it to toggle the LED state in a
     // loop.
@@ -81,7 +81,7 @@ fn main() -> ! {
             color.hue = hue;
             // Convert from the HSV color space (where we can easily transition from one
             // color to the other) to the RGB color space that we can then send to the LED
-            data = [hsv2rgb(color)];
+            data = [hsv2rgb(color); 2];
             // When sending to the LED, we do a gamma correction first (see smart_leds
             // documentation for details) and then limit the brightness to 10 out of 255 so
             // that the output it's not too bright.

```

### Expected

The on-board led should show a rainbow pattern, just like before.

### Actual

The led keeps the color that was active before the start of the program (no updates get through, not even the first one).

As far as I can see, the call to `SmartLedAdapter.write()` never returns. I haven't found the root cause for this yet.

Does anyone have an idea what could be going wrong?

fkohlgrueber, 2022-07-26:
Looking a bit further into this, it seems like `PulseControl` doesn't handle large inputs properly. Everything works fine if I send up to 47 pulses (with the 48th being zero / "end of transmission"). As soon as I'm using 48 or more pulses before I send the end marker, it seems like the first 48 pulses are sent repeatedly and the transmission never ends. Can anyone confirm this observation?

bjoernQ, 2022-07-27:
Thanks for opening this issue

@jessebraham since I think you are currently looking into `PulseControl` for other reasons maybe you want to have a look at this, too?

PR #143 (2022-08-09): Make RMT TX larger than one block work on ESP32-C3 and ESP32-S3
For ESP32-C3 and ESP32-S3 it wasn't possible to use RMT/PulseControl to transmit anything that is larger than one RMT memory block. This is a fix for that and also adds examples of using the raw PulseControl w/o the SmartLED feature.

This fixes #120 

Please note: because of the way how this works you have to compile in release mode if the whole transaction doesn't fit into one memory block
closed 2022-08-09 by jessebraham