esp-rs/esp-hal #123: SYSTIMER should support period mode

bjoernQ, 2022-07-27:
Currently we only support target mode.

Period mode could be potentially useful e.g. `esp-wifi` currently uses period mode and since the HAL lacks that functionality it accesses the registers directly.

MabezDev, 2022-11-15:
@bjoernQ I think this is already done for TIMG timers? When used with the e-h timer traits they are used in periodic mode? Please correct me if I'm wrong :D.

bjoernQ, 2022-11-15:
🤔 seems you are right 👀  .... but probably it shouldn't be like that? 

i.e. by default, it should be one-shot and we should have an "into_periodic" function (like systimer) to turn it into periodic mode?

bjoernQ, 2022-11-15:
Sounds like the best way to go - then it's aligned with systimer 👍

bjoernQ, 2022-11-15:
Ohh you are right

closed 2022-11-15 by MabezDev