esp-rs/esp-hal #127: How to specify dependency on esp-hal-common outside of esp-hal crate?

jeancf, 2022-07-31:
I wanted to play around with interrupt timers on esp32c3 and I bumped into an issue with the dependency on esp-hal-common.

Following the _timer_interrupt.rs_ example in esp32c3-hal I need to `use` `esp-hal-common::Timer`. This example being located inside the `esp-hal` crate, a path dependency can be easily specified (from Cargo.toml):

```
[dependencies.esp-hal-common]
path     = "../esp-hal-common"
features = ["esp32c3"]
```

However, from my app that is a different crate, how can I specify this dependency?

Wouldn´t if make sense to have `esp32c3-hal`(and the other hal) re-export `esp-hal-common`? The user does not need to know about the existence of `esp-hal-common` I guess.

bjoernQ, 2022-07-31:
It should already be re-exported as timer::Timer - if not it's wrong since you are right about esp-hal-common being an implementation detail.

In any case it's certainly bad to use esp-hal-common in the examples 

If you want to add the dependency it works exactly as for esp32c3-hal just changing the name

jeancf, 2022-07-31:
It is indeed, my bad. I had a typo in my `use` statement. Thanks.

closed 2022-07-31 by jeancf