esp-rs/esp-hal #142: [ESP32C3] unexpected behaviour of GPIO as input pin with weak pull-down resistor

jeancf, 2022-08-09:
I want to configure an input pin on an esp32c3 with a weak pull-down resistor so that when the pin is floating the input signal is registered as *low*.

I tested GPIO8 with 2 different methods (using HAL and manipulating registers) and with 2 different boards (AI thinker and Lilygo) with the same result: **the pin is read *high* when it is not connected.

I expect it to be *low* when unconnected or connected to ground and *high* only when it is connected to VCC.

## With HAL

```rust
let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
let mut echo = io.pins.gpio8.into_pull_down_input();
echo.enable_input(true);
```

Then I check the status of the pin with `echo.is_high().unwrap()` and `echo.is_low().unwrap()`.

## With register manipulation

```rust
peripherals.IO_MUX.gpio[8].write(|w| {
    w.fun_wpd().set_bit()
     .fun_wpu().clear_bit()
     .fun_ie().set_bit()
    });
```

Then I check the status of the pin with `peripherals.GPIO.in_.read().data_next().bits() & (0b1 << 8)) >> 8 == 1`

**What am I missing? Do I need to configure something else?**

Here is the [test](https://gitlab.com/jeancf/esp32c3-rust-pulldown-input/-/tree/main) I ran.

## UPDATE

**If I run the same test case with GPIO0 instead of GPIO8, the behavior is correct**

So I checked all pins in a [separate branch of the repo](https://gitlab.com/jeancf/esp32c3-rust-pulldown-input/-/tree/all_pins) and found out that pins 0-7 behave as expected but a number of others don´t (e.g. 8, 9, 10).

bjoernQ, 2022-08-09:
We merged a fix yesterday - e.g. GPIO 10 should now work as expected.

GPIO 8 however is a strapping pin - dev boards have a pull-up there, e.g. https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITM-1_V1_20200915A.pdf

However, if with the latest commits GPIO 10 still doesn't work it's probably still a bug (or those dev boards pull them high)

jeancf, 2022-08-09:
GPIO10 seems to be fine now after the latest changes. However some other are still not OK. Notably, GPIO6 was low when unconnected before the update, it is now high. 

Here is the latest overview on 2 different boards when trying to configure as input pull-down:

| GPIO   | AI Thinker | LILYGO |
| ------ | ---------- | ------ |
| GPIO0  | Low        | Low    |
| GPIO1  | Low        | Low    |
| GPIO2  | Low        | Low    |
| GPIO3  | Low        | Low    |
| GPIO4  | Low        | Low    |
| GPIO5  | Low        | Low    |
| GPIO6  | High       | High   |
| GPIO7  | Low        | Low    |
| GPIO8  | High       | High   |
| GPIO9  | High       | High   |
| GPIO10 | Low        | Low    |
| GPIO11 | High       | High   |
| GPIO12 | CRASH      | Low    |
| GPIO13 | Low        | Low    |
| GPIO14 | CRASH      | CRASH  |
| GPIO15 | CRASH      | CRASH  |
| GPIO16 | CRASH      | CRASH  |
| GPIO17 | CRASH      | CRASH  |
| GPIO18 | Low        | Low    |
| GPIO19 | High       | High   |
| GPIO20 | CRASH      | Low    |
| GPIO21 | CRASH      | CRASH  |

> *CRASH* means that the host stops receiving serial messages. I don´t know if the MCU actually crashes.

Of course, some GPIO are dedicated to other functions (8 as you note, but also 18/19 for JTAG, 20/21 for UART, etc.). But I would definitely expect 6 and 12 to be low when unconnected and in pull-down configuration.

bjoernQ, 2022-08-09:
Thanks for the detailed analysis! I happen to also own an AI Thinker dev-board and checked what ESP-IDF does: 
GPIO 6 is indeed low when configured as pull-down input _but_ for me it's also the case with `esp-hal`

None of my boards like GPIO12 / GPIO13 to be used (i.e. same as for you on the AI Thinker - it's used for flash on my boards)

GPIO 11 is also high for me (also with ESP-IDF)

So, I really wonder why GPIO6 is High on the AI Thinker board for you. Do you have any way to double check / verify that with e.g. ESP-IDF, MicroPython, Rust-Std or something else?

jeancf, 2022-08-09:
> So, I really wonder why GPIO6 is High on the AI Thinker board for you

So do I. This is only since this morning, presumably with the latest changes to esp32c3-hal. Yesterday it was low on both boards. I will check with ESF-IDF.

closed 2022-08-09 by jessebraham

ProfFan, 2023-11-01:
@bjoernQ I believe I have the same issue on the ESP32-S3