esp-rs/esp-hal #144: [ESP32-C3] Direct boot linker bug

MabezDev, 2022-08-09:
I can't seem to pin it to a specific example, as on two different machines I encounter the error on a different example (timer_interrupts & serial_interrupts). To get this output I ran the following:

```
cargo build --features direct-boot --examples --release
```

```
= note: rust-lld: warning: address (0x42000008) of section .text is not a multiple of alignment (256)
          rust-lld: error: section .bss load address range overlaps with .rwtext
          >>> .bss range is [0x42C0, 0x42CF]
          >>> .rwtext range is [0x42C8, 0x498F]
```

@bjoernQ I believe you worked on the DB feature/linker script, any ideas?

bjoernQ, 2022-08-09:
Oh ... will try to reproduce this myself
Don't have any idea right now.

Just a shot in the dark:
Something like changing ` .rwtext ORIGIN(REGION_RWTEXT) + _data_size : AT(_text_size + _rodata_size + _data_size){` to ` .rwtext ORIGIN(REGION_RWTEXT) + _data_size : AT(_text_size + _rodata_size + _data_size + SIZEOF(.bss)){`

But really just guessing

MabezDev, 2022-08-09:
That seems to fix it for me!

MabezDev, 2022-08-09:
Thinking on it though, this isn't really right because bss isn't loaded. Maybe we need to move bss section to after .rwtext?

bjoernQ, 2022-08-09:
Would probably be the better option 🤔worth a try

gustavonihei, 2022-08-09:
> Oh ... will try to reproduce this myself Don't have any idea right now.
>
> Just a shot in the dark: Something like changing ` .rwtext ORIGIN(REGION_RWTEXT) + _data_size : AT(_text_size + _rodata_size [...]

I believe this is the actual solution, because both BSS and the RWTEXT refer to the same area on the address space, although they are accessible from different buses.

The #49 PR required a similar solution, since both Direct Boot and MCUboot images follow a different image format than the one expected by IDF Bootloader.

PR #145 (2022-08-10): include bss size when calculating rwtext address
Closes #144
closed 2022-08-10 by bjoernQ