esp-rs/esp-hal #150: [ESP32C3] _embedded_hal_digital_v2_InputPin>::Error` doesn't implement `Debug`

jeancf, 2022-08-16:
I am trying to understand how to build an embedded-hal generic driver. I wrote a simple interface for an ultrasonic sensor on the ESP32C3.

The initialization takes one output GPIO and one input GPIO. I specify them as generics,  bound to implementing the `InputPin` and `OutputPin` traits of the *embedded-hal* crate. I compile the *embedded-hal* dependency with the `unproven` feature to get access to the v2 traits.

```rust
use embedded_hal::digital::v2::{InputPin, OutputPin} ;

pub struct HcSr04<'l, OP, IP> {
    trigger: &'l mut OP,
    echo: &'l IP,
}

impl<'l, OP: OutputPin, IP: InputPin> HcSr04<'l, OP, IP> {
    pub fn new(trigger: &'l mut OP, echo: &'l IP) -> HcSr04<'l, OP, IP> {
        Self { trigger, echo }
    }
    etc.
}
```

In my app I do this based on *esp32c3-hal*:

```rust
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let mut trigger = io.pins.gpio6.into_push_pull_output();
    let echo = io.pins.gpio7.into_pull_down_input();

    let mut hc_sr04 = hc_sr04_embedded_hal::HcSr04::new(&mut trigger, &echo);
```

But when I try to use the method `is_high()` of the *embedded-hal* `InputPin` trait the compiler complaints that:

```
error[E0277]: `<IP as _embedded_hal_digital_v2_InputPin>::Error` doesn't implement `Debug`
    --> src/lib.rs:70:35
     |
70   |         while self.echo.is_high().unwrap() {
     |                                   ^^^^^^ `<IP as _embedded_hal_digital_v2_InputPin>::Error` cannot be formatted using `{:?}` because it doesn't implement `Debug`
     |
     = help: the trait `Debug` is not implemented for `<IP as _embedded_hal_digital_v2_InputPin>::Error`
note: required by a bound in `Result::<T, E>::unwrap`
    --> /home/jeancf/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/result.rs:1103:12
     |
1103 |         E: fmt::Debug,
     |            ^^^^^^^^^^ required by this bound in `Result::<T, E>::unwrap`
help: consider further restricting the associated type
     |
55   |     pub fn measure(&mut self) -> Result<f32, HcSr04Error> where <IP as _embedded_hal_digital_v2_InputPin>::Error: Debug {
     |                                                           +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
```

This is strange because I see in esp-hal-common that `InputPin` and `OutputPin` declare `type Error = Infallible` and I know that `core::convert::Infallible` implements `Debug`.

How can I eliminate this error?

MabezDev, 2022-08-16:
This is actually nothing to do with esp-hal, but actually with your generic implementation :).

To get this to work, you need to tell rust that the error type implemented debug, this is needed because `unwrap()` requires it. You should be able to do this by introducing a new Error generic type on the impl, then saying that the error type for `IP` is the new error type. Something like `IP<Error = E>`.

jessebraham, 2022-08-16:
Closing this as it's not related to `esp-hal`, as explained by @MabezDev. Please feel free to continue discussing, of course.

closed 2022-08-16 by jessebraham

jeancf, 2022-08-17:
Thank you @MabezDev.

> you need to tell rust that the error type implemented debug

Indeed, this was the problem. I just modified the `impl` statement like so:

```rust
impl<'l, OP: OutputPin, IP: InputPin> HcSr04<'l, OP, IP>
where
    IP::Error: Debug,
    OP::Error: Debug,
```

It reassured the compiler that the Error type of the `IP` and `OP` trait objects that will be provided by the user does indeed implement Debug.