esp-rs/esp-hal #156: SPI malfunction since #101

AnthonyGrondin, 2022-08-19:
I'm using a card reader plugged on SPI and https://github.com/esp-rs/esp-hal/commit/2fe27536aac810b74e96aa651ff14d98a23ab7fc seems to have introduced some functionality issues.

For example, the reader has a version register, which outputs the version, and when compiling with
```toml
[profile.dev]
debug = true
opt-level = "z"
```
I can read the correct version (0x92), but when compiling with
```toml
[profile.dev]
debug = true
opt-level = "s"
```
I get a wrong value (0x14).

Also, the reader doesn't detect the card anymore.
I tested using the previous commit: https://github.com/esp-rs/esp-hal/commit/6e037b08cab7dec6ed073216ac80d555c8ce8871, and all functionalities work perfectly.

I'm using an ESP32S3.

AnthonyGrondin, 2022-08-19:
Reverting those two functions, to the previous commit (6e037b08cab7dec6ed073216ac80d555c8ce8871) seems to fix the bug.

https://github.com/esp-rs/esp-hal/blob/6e037b08cab7dec6ed073216ac80d555c8ce8871/esp-hal-common/src/spi.rs#L453-L460

https://github.com/esp-rs/esp-hal/blob/6e037b08cab7dec6ed073216ac80d555c8ce8871/esp-hal-common/src/spi.rs#L512-L553

AnthonyGrondin, 2022-08-19:
This section:
https://github.com/esp-rs/esp-hal/blob/ede5007f71fdeeb7369a6ddbee6a202e2f4c8837/esp-hal-common/src/spi.rs#L585-L589

Looks like to be the issue, reverting it to
https://github.com/esp-rs/esp-hal/blob/6e037b08cab7dec6ed073216ac80d555c8ce8871/esp-hal-common/src/spi.rs#L534-L536

appears to fix the issue.

jrmoulton, 2022-08-19:
That's really strange. When calling the transfer function the flush function should still be called at least once...

bjoernQ, 2022-08-22:
I tried to reproduce the problem with the `spi_loopback` example on ESP32-S3 but according to my logic analyzer it's all fine con current `main` branch - can you reproduce it with the `spi_lookback` example?

PR #165 (2022-08-24): Fix embedded-hal 0.2.x SPI implementation
While for embedded-hal 1.0.0 it's allowed to return before the operation is finished that is NOT true for embedded-hal 0.2.x

The implementation was missing a call to `flush`

This fixes #156

@AnthonyGrondin you might want to try this branch so we are sure it will actually fix #156
closed 2022-08-24 by jessebraham