esp-rs/esp-hal #172: Problem with UART1 on ESP32C3

darksv, 2022-08-28:
Hello. I'm trying to configure the second UART on ESP32C3 with this crate, but unfortunately it doesn't work - there is no data being transmitted according to the logic analyzer. I'm using `advanced_serial` example from this repo. I've tried various pin configurations, but only UART0 seemed to operate as intended. After switching to `esp-idf-hal` the UART1 worked correctly.

I appreciate any suggestions.

bjoernQ, 2022-08-29:
Thanks for opening this issue. I just double checked with the `advanced_serial` example for the ESP32-C3 that it still works for me which it does. The example itself works as expected and also my logic analyzer shows it's working as expected.

Since it works for you with `esp-idf-hal` we can probably rule out a hardware problem. Which dev-board are you using and which pin combinations did you try? (It should _just work_ with the pins in the example)

darksv, 2022-08-29:
I have ESP-C3-32S boards from Ai-Thinker - tried two of them.
Basically I used the example as-is, so these are pins 1 2. I also tried 4 5, 18 19 ... and probably some other pins and combinations as well including swapping pins for TX/RX.

bjoernQ, 2022-08-29:
That is odd - I also have those boards here and it works for me on them - 3,4 and 5 are connected to the onboard RGB LED so it might be a problem to use them but 1 + 2 definitely works for me

darksv, 2022-08-29:
I have checked everything again and it turned out that I messed up in `Cargo.toml` so that it was using IDF dependencies alongside with `esp-hal`. Everything works now. Sorry for bothering you. And thank you for verifying that the example indeed works!

closed 2022-08-29 by darksv