esp-rs/esp-hal #174: Add support for other clock sources in timer driver

JurajSadel, 2022-08-30:
Currently, we are using only `apb_clock` in General-purpose timers implementation. The https://github.com/esp-rs/esp-hal/pull/77 added other options for the clock source.

JurajSadel, 2022-08-31:
After discussion with @bjoernQ it seems it's not that straightforward. 

In ESP-IDF the user can set the `clock` source but also manually choose the divider etc - it's a much [lower level API](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-reference/peripherals/timer.html). 
Do we want to have an additional low-level constructor that takes the clock source and divider? Then the user will want to use `ticks` for e.g start function - probably going to be a lot of effort.
 Also would be a general design decision since if we offer a low-level version here why don't we have it everywhere? 
 
 For now, let's turn this issue into a discussion of how we want to deal with this in the future.
 What is your opinion @bjoernQ, @jessebraham, @MabezDev?

MabezDev, 2024-02-19:
ABP has served us well and I'm not aware of any current issues with TIMG being driven from it, therefore I'm going to close this for now.