esp-rs/esp-hal #175: Lack of uninit section causes linking defmt to fail

9names, 2022-08-30:
defmt puts it's uninitialised data in the uninit section, found here in the cortex-m-rt linker script:
https://github.com/rust-embedded/cortex-m/blob/0e530549de322684c50e858c6bb985afb5479dbe/cortex-m-rt/link.x.in#L170

this is absent from the linker scripts here, so we get this linker error:
```console
note: rust-lld: warning: address (0x42000008) of section .text is not a multiple of alignment (256)
          rust-lld: error: section .uninit.defmt-rtt.BUFFER load address range overlaps with .rwtext
          >>> .uninit.defmt-rtt.BUFFER range is [0x2D4E4, 0x2D8E3]
          >>> .rwtext range is [0x2D4EC, 0x2DDFB]
```
This issue has already been reported with defmt.
https://github.com/knurling-rs/defmt/issues/693

I've tested adding this uninit section to the esp32c3 db-riscv-link.x after BSS, and it seems to work!
```
  /* ### .uninit */
  .uninit (NOLOAD) : ALIGN(4)
  {
    . = ALIGN(4);
    __suninit = .;
    *(.uninit .uninit.*);
    . = ALIGN(4);
    __euninit = .;
  } > REGION_BSS
```

Also seems to trigger some issues flashing on cargo embed, but I think those are likely to be probe-rs bugs.
probe-rs-cli does not trigger this, maybe my cargo-embed is not up to date. Recording it here just in case:
```
Error failed to flash blinky
Caused by:
                 0: Error while flashing
                 1: No flash memory contains the entire requested memory range 0x00000000..       0x8.
```

I'll make a PR for this in a few days if no-one else picks this up

TheButlah, 2022-09-04:
Anything else that needs to be done in this PR?

9names, 2022-09-04:
No, that should be all that is necessary.
I only tested it on the direct-boot linker script, you should probably test against the regular linker script too.
I assume if it works for esp32c3 it should be safe to put in all the other chip variants as well.

TheButlah, 2022-09-04:
I did not have any issues flashing with `cargo embed`. However, my rtt console looks like this:
[image]

Code is [here](https://github.com/SlimeVR/SlimeVR-Rust/blob/7d279d3ccef390502cd5bcd62afe729fe64ba43e/firmware/src/main.rs)

I tried to use your linker script patch like this:
`db-riscv-link.x`:
```
  .bss (NOLOAD) :
  {
    _sbss = .;
    *(.sbss .sbss.* .bss .bss.*);
    . = ALIGN(4);
    _ebss = .;
  } > REGION_BSS

   /* ### .uninit */
  .uninit (NOLOAD) : ALIGN(4)
  {
    . = ALIGN(4);
    __suninit = .;
    *(.uninit .uninit.*);
    . = ALIGN(4);
    __euninit = .;
  } > REGION_BSS

  /* fictitious region that represents the memory available for the heap */
  .heap (NOLOAD) :
  {
    _sheap = .;
    . += _heap_size;
    . = ALIGN(4);
    _eheap = .;
  } > REGION_HEAP
```

TheButlah, 2022-09-04:
My apologies, this was user error. I forgot that `defmt` sends compressed data, so merely printing it without decoding it will look like spaghetti.

I have successfully tested the `direct-boot` approach, will test the regular boot one soon.

PR #179 (2022-09-07): Fixed defmt on direct-boot mode
Closes: #175 

Ideally, I would also like to add this for the regular boot mode. I have not tested that yet.

Author is @9names, I merely submitted the PR.
closed 2022-09-07 by jessebraham