esp-rs/esp-hal #177: Could not compile 'esp32c3-hal due to 550 previous errors

dcaponi, 2022-09-01:
Hey Team, 

I tried running one of the examples (blinky) and the crate doesn't compile due to what looks to be 550 syntax errors.

Here's my `cargo.toml`

```
[package]
name = "blinky"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
esp32c3-hal = "0.1.0"
```

This is the first of those errors that gets logged out 

```
esp32c3-hal-0.1.0/src/gpio.rs:12:1
   |
12 | / gpio! {
13 | |     Function1,
14 | |     SingleCore,
15 | |
...  |
37 | |     Gpio21: (gpio21, 21, gpio[21], IO,   0, Bank0, None), (), (U0TXD: Function0),
38 | | }
   | |_^
   |
   = note: this error originates in the macro `impl_input` (in Nightly builds, run with -Z macro-backtrace for more info)
```

Any advice for troubleshooting or pointing me in the right direction if I'm doing something wrong would be greatly appreciated.

dcaponi, 2022-09-01:
Update - I think this might be caused by a nested dependency somewhere. If I import the crate using a git link i.e `esp32c3-hal = { git = "https://github.com/esp-rs/esp-hal.git", features = ["ufmt"] }` and build it works fine. Inspecting the cargo.lock file I see Paste = "1.0.8". 

If I try using the crate straight from crates.io i.e. esp32c3-hal = "0.1.0" I get the aforementioned error and upon inspecting cargo lock I see Paste = "1.0.9". That is literally the only difference. I think Fugit, which uses GCD which is pegging Paste at ^1 might be importing some breaking code somewhere.

If anyone else is having / able to reproduce this issue I might file a PR with GCD to peg the version of Paste at 1.0.8. I don't really know how to explain the issue to the Paste folks to submit a PR or raise an issue there as macros are kind of foreign to me 😢 

TLDR as a possible workaround for this issue for anyone who runs into it, just use the git version `esp32c3-hal = { git = "https://github.com/esp-rs/esp-hal.git", features = ["ufmt"] }` as a workaround due to some dependency hell issue

jessebraham, 2022-09-01:
Unfortunately `paste` had a patch release (`1.0.9`) which resulted in an error on our end. I'm not sure how this happened exactly but that's macros for you.

The fix for now is pinning to `=1.0.8` for `paste`. I'm working on a fairly extensive rework of the `gpio` module which should resolve this, I'm hoping to get this completed soon so we can publish a release with the fix.

dcaponi, 2022-09-01:
Sounds good thanks for the update. Glad it wasn't just be doing some bone-head thing.

jessebraham, 2022-09-13:
We have published new versions of each HAL, so this should be resolved when using `esp32c3-hal 0.2.0`.

closed 2022-09-13 by jessebraham