esp-rs/esp-hal #180: Gapless SPI transfers

bjoernQ, 2022-09-05:
Currently our code transfers SPI data in chunks (sized by `FIFO_SIZE`) . This results in short gaps in the transmission.
While most (if not all) SPI devices should work this way this prevents us from (mis)using SPI to do other things like driving an LED-strip of addressable LEDs or similar.

It is possible to continuously write to the FIFO and avoid those gaps. Something like that is already done for `pulse_control`/`RMT`

dougli1sqrd, 2022-09-29:
Oh is there a way to submit data to the SPI fifo directly? Is your ticket referring to when performing CPU controlled SPI transfers there seems to be large sets of time where SPI is not transferring data? I have a project where I'm trying to pump out data over quad spi as fast as I can, and would definitely be interested in seeing ways of improving speed.

bjoernQ, 2022-09-29:
Yes that might be possible  - from the ESP32-C3 TRM (it's similar for the others)

> 25.5.5.1 CPU­Controlled Master Mode
> ... . If the data
byte length is larger than 64, the data in SPI_W8_REG[7:0] ~ SPI_W15_REG[31:24] may be sent more
than once. For instance, 66 bytes (byte0 ~ byte65) need to send out, the address of byte65 is the
result of (65 % 64 = 1), i.e. byte65 is from SPI_W0_REG[15:8], and byte64 is from SPI_W0_REG[7:0].
For this case, the content of SPI_W0_REG[15:0] may be sent more than once.

Unfortunately, there doesn't seem to be a threshold interrupt or anything - so it might be tricky.  Also it's not clear if the hardware takes writes to the SPI_Wx_REGs during a transfer.

Probably this is more about researching this.

If it turns out we can't get it working we probably need to wait for the implementation of DMA (which we definitely want to do - it's just not yet started)

dougli1sqrd, 2022-09-29:
oh interesting, like maybe you could write into the Wx registers and perform a long write with SPI and then "chase" the registers. I wonder when one of  the SPI_Wx_REG is read by SPI does it empty? 

There is also
```
// wait if currently in a write
            while block.cmd.read().usr().bit_is_set() {
                // wait
            }
```
but I guess that's just for "is SPI currently writing".

sethp, 2022-09-29:
>  Also it's not clear if the hardware takes writes to the SPI_Wx_REGs during a transfer.

I ran a small experiment that set up a 128-byte transfer (in quad mode) and flipped the `Wx` registers from `0x0000_0000` to `0xffff_ffff` while the transfer was running, and the good news is that the hardware does "see" the writes:

[image]

As you say, though, it does seem tricky to get the timing right: in that screenshot, it looks like I overwrote the back ~half of the Wx block before it was sent the first time. Letting the analyzer run for a while and eyeballing it, though, it does look like the rising edge there is consistent, which leads me to believe it'd be possible to emit a very long (though not fully continuous) waveform this way.

bjoernQ, 2022-10-12:
FYI I started implementing DMA (for the start on ESP32-C3). With that we can have single transfers up to 32k which is probably the way to go

closed 2022-10-31 by bjoernQ