esp-rs/esp-hal #188: Fails to link without importing prelude

JMS55, 2022-09-14:
1. Generate a new project for ESP32-C3
2. Go to main.rs, and remove the prelude import: `use esp32c3_hal::prelude::*;`
3. `cargo build`
`rust-lld: error: [PROJECT]/target/riscv32imc-unknown-none-elf/debug/build/esp32c3-hal-dbfe20a1aa708601/out/bl-riscv-link.x:2: symbol not found: _start_trap_hal`

bjoernQ, 2022-09-14:
Thanks for creating this issue

Unfortunately, I cannot reproduce this. What I did

- `cargo generate https://github.com/esp-rs/esp-template` 
- in `main.rs` remove the import of the prelude and import the now missing imports (`system::SystemExt`, `embedded_hal::watchdog::WatchdogDisable` ... embedded-hal also needs to be added as a dependency)
- `cargo build`

Maybe I missed something from your description.

Tested on Windows 11 with rustc 1.65.0-nightly (1120c5e01 2022-09-08)

MabezDev, 2022-09-14:
This probably isn't specific to the prelude, but the fact that esp32c3_hal needs to be included in the binary to pull in the `_start_trap_hal` etc. If you could share your whole project it might be easier to see whats going on.

JMS55, 2022-09-14:
Ah, I had removed everything in main.rs to get a minimal program going, without any functionality. So I didn't use any part of the hal crate, and didn't have that symbol then.

MabezDev, 2022-09-14:
Yes that'll be it! Glad we got to the bottom of it :)

closed 2022-09-14 by MabezDev