esp-rs/esp-hal #190: Persistent flash storage or FS

D1plo1d, 2022-09-15:
Is there any way to read and write persistent data to the flash memory from esp-hal? It looks like the esp32c3-wroom has 4MB of external flash which could be great for persisting small files at runtime if there's room left over after storing the application itself.

I'm imagining something akin to either:
- A high-level key/value store like: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html
- Or a low-level SPI flash API like: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html

jessebraham, 2022-09-15:
@bjoernQ has created the [esp-storage](https://github.com/bjoernQ/esp-storage) repository for this purpose, however I'm not sure of its current state and/or capabilities. Might be a start though, this isn't really territory we've explored in this project yet.

bjoernQ, 2022-09-15:
`esp-storage` should work fine for all four supported chips now - it's a more low-level API on which one could build e.g a KV store, a file-system etc.

Since `esp-storage` implements embedded-storage such an implementation wouldn't even be tied to our chips

That said - I will move `esp-storage` to the `esp-rs` organization soon and also publish it on crates.io, then

PR #199 (2022-09-28): Add a section to the README listing some other useful crates
Figured it was useful to at least list some of these crates in the top-level README. I've excluded `esp-wifi` for the time being as it has not yet been published.

Any objections? Additional crates we should mention?

Closes #190 too, I guess.
closed 2022-09-28 by jessebraham