esp-rs/esp-hal #197: Improve easy-of-use for `#[interrupt]` macro on ESP32-C3

bjoernQ, 2022-09-27:
The macro needs to reference the TrapFrame ( https://github.com/esp-rs/esp-hal/blob/7889a992d7efcbbba0a5c7ef679694877f30380e/esp-hal-procmacros/src/lib.rs#L224 )

This requires importing something like `use esp32c3_hal::interrupt;` in the root

Maybe we can come up with a better solution which doesn't require that import

MabezDev, 2022-09-30:
Yeah, I hate that part, it's super annoying. For context, this "check" is here to make sure a given interrupt actually exists. Hopefully we can find a better way to do this check.

dimpolo, 2022-12-07:
Same thing applies to this line requiring `pac` to be in scope:
https://github.com/esp-rs/esp-hal/blob/4ab05e89235e620287a6c3b24486c0b8c2c05a28/esp-hal-procmacros/src/lib.rs#L202

MabezDev, 2022-12-22:
I don't think there is a way around having to import the TrapFrame, because its part of the public API of the function. Thankfully, because it's public it's easy to spot that it needs to be imported.

> Same thing applies to this line requiring pac to be in scope:

I've been looking at improving this, I don't think there is an easy way to check, without doing strange things like having `const` `to_string` methods on each of the interrupt enums and comparing the to string with the name of the interrupt handler.

I'm thinking about removing the check entirely, and instead adding a log statement inside each hal's `extern "C" fn EspDefaultHandler(_interrupt: peripherals::Interrupt) {}` saying "Unhandled Interrupt : X". Thoughts?

PR #391 (2023-02-10): Remove unnecessary `rt` crate dependencies
Fixes #382 

- remove dependencies to `rt` crates and `riscv` / `xtensa-lx` from the chip specific HALs, re-export from esp-hal-common
- re-export `riscv` / `xtensa-lx` in the chip specific HALs
- `entry` is re-exported and added to the prelude
- atomic emulation is always included by the HAL for ESP32-S2 (like it's already the case for ESP32-C2 / ESP32-C3)

This PR looks huge but most of the changes are in the examples.

Nice side-effect: fixes #197 

This has some potential for bikeshedding about some names (e.g. `trapframe::TrapFrame`) - I tried to [... cut]