esp-rs/esp-hal #200: esp-hal-common fails to build on crates.io

jeancf, 2022-09-28:
I don't know if you are aware but esp-hal-common fails to build on crates.io. As a result the documentation is also unavailable on docs.rs. 

https://docs.rs/crate/esp-hal-common/latest has some details:
```
# rustc version
rustc 1.65.0-nightly (fa6ee9375 2022-09-12)
# docs.rs version
docsrs 0.6.0 (c4f9e90 2022-09-04)

# build log
[INFO] running `Command { std: "docker" "create" "-v" "/home/cratesfyi/workspace/builds/esp-hal-common-0.2.0/target:/opt/rustwide/target:rw,Z" "-v" "/home/cratesfyi/workspace/builds/esp-hal-common-0.2.0/source:/opt/rustwide/workdir:ro,Z" "-v" "/home/cratesfyi/workspace/cargo-home:/opt/rustwide/cargo-home:ro,Z" "-v" "/home/cratesfyi/workspace/rustup-home:/opt/rustwide/rustup-home:ro,Z" "-e" "SOURCE_DIR=/opt/rustwide/workdir" "-e" "CARGO_TARGET_DIR=/opt/rustwide/target" "-e" "DOCS_RS=1" "-e" "CARGO_HOME=/opt/rustwide/cargo-home" "-e" "RUSTUP_HOME=/opt/rustwide/rustup-home" "-w" "/opt/rustwide/workdir" "-m" "3221225472" "--cpus" "3" "--user" "1001:1001" "--network" "none" "ghcr.io/rust-lang/crates-build-env/linux@sha256:824c28ce115f6b999260af6986f3384c158e782489540e41c0b698ca1d9fd927" "/opt/rustwide/cargo-home/bin/cargo" "+nightly" "rustdoc" "--lib" "-Zrustdoc-map" "-Z" "unstable-options" "--config" "build.rustdocflags=[\"-Z\", \"unstable-options\", \"--emit=invocation-specific\", \"--resource-suffix\", \"-20220912-1.65.0-nightly-fa6ee9375\", \"--static-root-path\", \"/\", \"--cap-lints\", \"warn\", \"--disable-per-crate-search\", \"--extern-html-root-takes-precedence\"]" "-Zunstable-options" "--config=doc.extern-map.registries.crates-io=\"https://docs.rs/{pkg_name}/{version}/x86_64-unknown-linux-gnu\"" "-j3" "--target" "x86_64-unknown-linux-gnu", kill_on_drop: false }`
[INFO] [stderr] WARNING: Your kernel does not support swap limit capabilities or the cgroup is not mounted. Memory limited without swap.
[INFO] [stdout] b08b92f11a64cf77130e7cd445c9bca664a5ce11aa64bf56ced3e0c42eb08c3c
[INFO] running `Command { std: "docker" "start" "-a" "b08b92f11a64cf77130e7cd445c9bca664a5ce11aa64bf56ced3e0c42eb08c3c", kill_on_drop: false }`
[INFO] [stderr]    Compiling esp-hal-common v0.2.0 (/opt/rustwide/workdir)
[INFO] [stderr]     Checking fugit v0.3.6
[INFO] [stderr]    Compiling quote v1.0.21
[INFO] [stderr] error: failed to run custom build command for `esp-hal-common v0.2.0 (/opt/rustwide/workdir)`
[INFO] [stderr] 
[INFO] [stderr] Caused by:
[INFO] [stderr]   process didn't exit successfully: `/opt/rustwide/target/debug/build/esp-hal-common-238d17f338d72b4e/build-script-build` (exit status: 101)
[INFO] [stderr]   --- stderr
[INFO] [stderr]   thread 'main' panicked at 'Exactly one chip must be enabled via its cargo feature', build.rs:10:9
[INFO] [stderr]   note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[INFO] [stderr] warning: build failed, waiting for other jobs to finish...
[INFO] running `Command { std: "docker" "inspect" "b08b92f11a64cf77130e7cd445c9bca664a5ce11aa64bf56ced3e0c42eb08c3c", kill_on_drop: false }`
[INFO] running `Command { std: "docker" "rm" "-f" "b08b92f11a64cf77130e7cd445c9bca664a5ce11aa64bf56ced3e0c42eb08c3c", kill_on_drop: false }`
[INFO] [stdout] b08b92f11a64cf77130e7cd445c9bca664a5ce11aa64bf56ced3e0c42eb08c3c
```

jessebraham, 2022-09-28:
Thanks for opening this issue. This is expected behaviour, the `esp-hal-common` crate requires one of the chip features to be activated in order to build (as stated by the error message in the build). The package is not meant to be used on its own, as everything relevant is re-exported by each HAL crate:

- https://docs.rs/esp32-hal/latest/esp32_hal/
- https://docs.rs/esp32c3-hal/latest/esp32c3_hal/
- https://docs.rs/esp32s2-hal/latest/esp32s2_hal/
- https://docs.rs/esp32s3-hal/latest/esp32s3_hal/

jeancf, 2022-09-28:
Ah sorry. I understand the error message now, thanks. I will clone the repo and generate the doc locally then.

closed 2022-09-28 by jeancf