esp-rs/esp-hal #201: calling `write_read` with an empty write buffer repeatedly retries I2C reads

TheButlah, 2022-09-28:
With this code on an ESP32-C3:

```rust
i2c
  .write_read(ADDR_DOESNT_MATTER, &[], &mut [0; RECV_LEN_DOESNT_MATTER])
  .expect("Failed i2c");
```

I get this signal trace: [repeated.zip](https://github.com/esp-rs/esp-hal/files/9668043/repeated.zip)

[image]

The code panics due to the `NACK` and then suspends execution in a loop - I have triple checked this with both `probe-rs` and `defmt` logs. So my only conclusion is that the hardware I2C is in a broken state that causes repeated retransmission.

Probably, it is due to this code in `esp-hal`:
[image]

The write command is issued, but the data to write is never actually set.

bjoernQ, 2022-09-29:
Thanks for opening this issue!

I did some tests on my side, too.

This is with a no bytes write and something on the bus that ACKs:
[image]

This is with a one bytes write and something that ACKs:
[image]

This is with a no bytes write and nothing that ACKs: (i.e. I think that is your case)
[image]

This is with a one byte write and nothing that ACKs:
[image]

So, what is repeated seems to be the attempt to read.

bjoernQ, 2022-09-29:
We can get rid of those read attempts by adding `self.reset();` to `execute_transmission` before returning `return Err(Error::AckCheckFailed);` (maybe we should do that in all error cases)

bjoernQ, 2022-10-04:
Just out of curiosity: is this related to your problem using MPU6050? I ordered a MPU6050 break-out board but which should hopefully arrive by the end of the week

TheButlah, 2022-10-04:
Yes, it was. I added [this](https://github.com/SlimeVR/SlimeVR-Rust/pull/36/files#diff-0d8effb99b25ba92abf10f742584b05ad99d057c8e6e05055ca8124dc0ee4289R27) PR in which originally I was writing an empty buffer as an attempt to reset any internal state in the I2C peripheral (because IDK what I'm doing lmao)

I now write a single `0x00` instead, and that results in successful communication with the MPU6050

jessebraham, 2022-11-09:
Can this PR be closed now that #233 has been merged?