esp-rs/esp-hal #224: Add support for the ESP32-H2

jessebraham, 2022-10-24:
I'll be taking care of this one as well, and this likely won't happen until the new year, but just wanted to create a tracking issue regardless.

### Status

- [ ] Create an SVD
- [ ] Create a PAC
- [ ] Add support to ancillary crates (`esp-backtrace`, `esp-println`)
- [ ] Create a HAL package

jessebraham, 2023-02-10:
I've begun work on HAL support for this device.

jessebraham, 2023-04-18:
Development will take place in this branch until the changes are ready to merge into `main`:  
https://github.com/esp-rs/esp-hal/tree/feature/esp32h2

jessebraham, 2023-05-09:
Initial support for this device has been merged into `main`. We will continue to add support for additional peripherals in the coming weeks.

SergioGasquez, 2023-05-11:
> Also, [I've tried to add support for RNG](https://github.com/SergioGasquez/esp-hal/tree/feature/esp32h2-rng) but the bootloader that we currently use does not have `bootloader_random_enable/bootloader_random_disable` implemented yet, hence [RNG can't work](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/random.html). We'll need to check if there is a more recent bootloader or wait until those methods are implemented.

Just tried with the latest esp-idf bootloader, and it still does not implement those methods, here is how I tried it:
1. Downloaded esp-idf master
2. Installed it for esp32h2
4. Built, flash and monitored (`idf.py flash monitor`) one of the examples (`get-started/blink`)

Results in:
```
I (23) boot: ESP-IDF v5.2-dev-503-g17451f1fb3 2nd stage bootloader
...
W (48) bootloader_random: bootloader_random_enable() has not been implemented yet
...
I (155) boot: Disabling RNG early entropy source...
W (160) bootloader_random: bootloader_random_disable() has not been implemented yet
```

jessebraham, 2023-05-17:
Added a few more examples in #548 (`advanced_serial`, `embassy_hello_world`, `systimer`, `usb_serial_jtag`)

SergioGasquez, 2023-05-18:
Working on `PCNT`!

SergioGasquez, 2023-05-22:
Working on `RMT`!

JurajSadel, 2023-05-25:
I will look into LEDC

SergioGasquez, 2023-05-30:
Im working in `debug_assist`

JurajSadel, 2023-05-30:
I'm working on ADC 😅

JurajSadel, 2023-05-31:
Looking into `timer_interrupt` example

jessebraham, 2023-05-31:
Will tackle `RMT` and the `hello_rgb` example next

JurajSadel, 2023-06-06:
I'll look into `i2s`

jessebraham, 2023-06-19:
With #597 merged we have completed all examples for this device!

We should done one final round of review (clean up any warnings, etc.) and verify that all examples are still working (this is in-progress) and then I think we're good for our first release!

Thanks to everybody who contributed to supporting this device!

jessebraham, 2023-07-04:
[esp32h2-hal@0.1.0](https://crates.io/crates/esp32h2-hal/0.1.0) has been published!

closed 2023-07-04 by jessebraham