esp-rs/esp-hal #237: GPIO Interrupt handler per GPIO

Miroito, 2022-10-29:
I'd like to use 2 different functions to handle 2 different gpio interrupts in a program using the esp32c3-hal.

I can't find a way to do this with this hal given that the Interrupt enum only contains a GPIO variant that doesn't care which pin triggered the interrupt.

Is there a prospect of supporting this functionality in this crate? Is there a workaround that I can use now?

The Pin trait doesn't allow it right now but it would be great to just be able give `listen` a function to run on the defined event.

bjoernQ, 2022-10-31:
The hardware really generates the one GPIO interrupt for all the pins.

What you can do is using https://docs.rs/esp32c3-hal/0.2.0/esp32c3_hal/gpio_types/trait.Pin.html#tymethod.is_pcore_interrupt_set to see which GPIO triggered the interrupt and call your function to handle it.

Technically we could do that in the HAL for convenience but it would add some latency for everyone

Miroito, 2022-11-03:
> The hardware really generates the one GPIO interrupt for all the pins.
This is were my knowledge of the esp32 was lacking, thanks for clearing this point.

> What you can do is using https://docs.rs/esp32c3-hal/0.2.0/esp32c3_hal/gpio_types/trait.Pin.html#tymethod.is_pcore_interrupt_set to see which GPIO triggered the interrupt and call your function to handle it.
Thanks, I completely missed this function probably because there is no description provided. I'm going to try it for sure.

> Technically we could do that in the HAL for convenience but it would add some latency for everyone
Given it is possible to check if the pin currently has an interrupt if needed, I see no point to implement extra latency into the HAL either. 

Thanks for the solution

closed 2022-11-03 by Miroito