esp-rs/esp-hal #249: Tracking the "next" branch

jessebraham, 2022-11-04:
This work is inspired largely by the same effort already undergone in `esp-idf-hal`, which in turn was based on the work done in `embassy`.

https://github.com/esp-rs/esp-idf-hal  
https://github.com/embassy-rs/embassy  

Development will be done in the `next` branch. I have added a single commit which adds some types, traits, and macros which will be required for development.

https://github.com/esp-rs/esp-hal/tree/next

This branch will be periodically rebased on `main` to keep it in sync as much as possible. As we're able we can merge changes into `main`, repeating this until the process is complete.

A number of peripherals will need their high-level interfaces re-written, though we can likely take this directly from `esp-idf-hal` with only minor modification. Hopefully most of the low-level code can be kept and/or adapted.

## Tasks

The tasks below do _not_ need to be completed in any particular order; however based on previous discussions, `UART` should be completed first and `GPIO` will be completed last.

- [ ] UART
- [ ] I2C
- [ ] SPI
- [ ] LEDC
- [ ] RMT
- [ ] GPIO

jessebraham, 2022-11-04:
I have previously done some work on both the `UART` and `GPIO` modules, and will get both projects cleaned up and pushed to branches soon.

jessebraham, 2022-12-12:
With #272 merged we have all the required building blocks to move forward with this, plus a `UART` implementation for reference.

bjoernQ, 2022-12-13:
For reference: #299 is a good example what needs to be done to convert a peripheral driver to the use the PeripheralRef pattern

MabezDev, 2022-12-13:
Working on SYSTIMER

jessebraham, 2022-12-13:
Working on I2C

SergioGasquez, 2022-12-13:
Working on USB

jessebraham, 2022-12-13:
Have `RNG` finished locally (will PR tomorrow) and then I'll tackle `SHA` after that.

bjoernQ, 2022-12-14:
Working on SYSTEM/DPORT

MabezDev, 2022-12-14:
Working on LEDC ☠️

bjoernQ, 2022-12-14:
Working on ClockControl

MabezDev, 2022-12-14:
Working on MCPWM

bjoernQ, 2022-12-14:
Working on DMA

bjoernQ, 2022-12-15:
Working on USB_OTG

bjoernQ, 2022-12-15:
Working on RTC

bjoernQ, 2022-12-15:
no idea what to do about TWAI since it's not merged yet 🤷‍♂️

bjoernQ, 2022-12-15:
Looking into GPIO

jessebraham, 2022-12-15:
> no idea what to do about TWAI since it's not merged yet 🤷‍♂️ 

I am going to try and get a test project set up for this next week so that we can get that PR merged.

jessebraham, 2022-12-19:
Have I missed any peripherals in the first post? I will go through the code base later, but wouldn't mind some more eyes on it just to be sure.

jessebraham, 2022-12-22:
`TWAI` PR has been merged, so I suppose we're done here 😁 Thanks for the help everybody!

closed 2022-12-22 by jessebraham