esp-rs/esp-hal #269: ESP32-S3 crashes when connected via Serial/Jtag port and sending two bytes of data

bjoernQ, 2022-11-21:
## How to reproduce

- Use a DevKit-C, connect USB to the USB port (i.e. not the USB UART)
- run the `usb_serial_jtag` example and see it continuously printing `Hello World!`
- send two characters e.g. in espflash monitor press a letter key twice
- now the code stops running

When that happens even CTRL-C / CTRL-R doesn't work anymore and you need to unplug the dev board

dimpolo, 2022-12-05:
I looked into this a bit.
The problem in the `usb_serial_jtag` example seems to have something to do with the interrupt not running correctly.
Receiving data outside of the interrupt seems to be working correctly.

bjoernQ, 2022-12-06:
Thanks for looking into this. One very interesting observation is that it seems to be different for different serial monitors - like on Windows `hterm` isn't even able to connect while with `putty` it doesn't seem to crash

bugadani, 2023-10-26:
I tried reproducing this issue on my custom esp32s3-based hardware but coudln't.

```
I (46) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (46) boot: compile time Jun  7 2023 08:07:32
I (47) boot: Multicore bootloader
I (51) boot: chip revision: v0.1
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (60) boot.esp32s3: SPI Mode       : DIO
I (65) boot.esp32s3: SPI Flash Size : 8MB
I (69) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 007f0000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=07044h ( 28740) map
I (128) esp_image: segment 1: paddr=0001706c vaddr=3fc89888 size=00004h (     4) load
I (129) esp_image: segment 2: paddr=00017078 vaddr=40378000 size=01888h (  6280) load
I (140) esp_image: segment 3: paddr=00018908 vaddr=00000000 size=07710h ( 30480)
I (154) esp_image: segment 4: paddr=00020020 vaddr=42000020 size=2448ch (148620) map
I (192) boot: Loaded app from partition at offset 0x10000
I (192) boot: Disabling RNG early entropy source...
Hello world!
Hello world!
Hello world!
USB serial interrupt
Read byte: 73
USB serial interrupt
Read byte: 73
Hello world!
Hello world!
USB serial interrupt
Read byte: 73
USB serial interrupt
Read byte: 64
USB serial interrupt
Read byte: 64
USB serial interrupt
Read byte: 73
USB serial interrupt
Read byte: 64
USB serial interrupt
Read byte: 61
USB serial interrupt
Read byte: 73
Hello world!
USB serial interrupt
Read byte: 64
USB serial interrupt
Read byte: 6c
USB serial interrupt
Read byte: 6a
USB serial interrupt
Read byte: 6c
USB serial interrupt
Read byte: 6b
```

bjoernQ, 2023-10-26:
Wow - a self-healing issue 🤣