esp-rs/esp-hal #277: Object safe error when making array of Gpio digital output pins

rathod-sahaab, 2022-11-28:
Hi! I am very familiar with c++, I am trying to implement a progress bar with LEDs where more can be added (numbers decided at compilation).

From my c++ thought process, I am trying to store pointers in an array derived from a virtual base class like OutputGpio and calling functions on them. I have tried to implement the same in rust. I guess I am expecting some wrong trait object

```rust
use esp32_hal::gpio_types::OutputPin;

pub struct GpioProgressBar<'a, const MAX_PINS: usize> {
    pins: [&'a dyn OutputPin; MAX_PINS],
}

impl<'a, const MAX_PINS: usize> GpioProgressBar<'a, MAX_PINS> {
    pub fn new(pins: [&dyn OutputPin; MAX_PINS]) -> Self {
        Self { pins }
    }
}

impl<'a, const MAX_PINS: usize> Progress for GpioProgressBar<'a, MAX_PINS> {
    fn show(&mut self, bounds: Bounds<usize>, current: usize) {
        let progress = (MAX_PINS * (current - bounds.min)) / (bounds.max - bounds.min);

        (0..MAX_PINS).for_each(|i| {
            if i < progress {
                self.pins[i].set_high().unwrap_or(());
            } else {
                self.pins[i].set_low().unwrap_or(());
            }
        })
    }
}
```
Trying to use it as

```rust
let mut led = io.pins.gpio25.into_push_pull_output();
let mut led2 = io.pins.gpio24.into_push_pull_output();

let mut progress_bar = GpioProgressBar::new([led, led2]);
```

Error:
```
error[E0038]: the trait `esp32_hal::gpio_types::OutputPin` cannot be made into an object
   --> src/ui/progress/gpio_progress_bar.rs:8:16
    |
8   |     pins: [&'a dyn OutputPin; MAX_PINS],
    |                ^^^^^^^^^^^^^ `esp32_hal::gpio_types::OutputPin` cannot be made into an object
    |
note: for a trait to be "object safe" it needs to allow building a vtable to allow the call to be resolvable dynamically;
```

rathod-sahaab, 2022-11-28:
Got something working, reviews appreciated, thanks!

```rust
use core::convert::Infallible;

use esp32_hal::prelude::_embedded_hal_digital_v2_OutputPin as OutputPin;

use crate::commons::bounds::Bounds;

use super::progress::Progress;

pub struct GpioProgressBar<'a, const MAX_PINS: usize> {
    pins: [&'a mut dyn OutputPin<Error = Infallible>; MAX_PINS],
}

impl<'a, const MAX_PINS: usize> GpioProgressBar<'a, MAX_PINS> {
    pub fn new(pins: [&'a mut dyn OutputPin<Error = Infallible>; MAX_PINS]) -> Self {
        Self { pins }
    }
}

impl<'a, const MAX_PINS: usize> Progress for GpioProgressBar<'a, MAX_PINS> {
    fn show(&mut self, bounds: Bounds<usize>, current: usize) {
        let progress = (MAX_PINS * (current - bounds.min)) / (bounds.max - bounds.min);

        (0..MAX_PINS).for_each(|i| {
            if i < progress {
                self.pins[i].set_high().unwrap_or(());
            } else {
                self.pins[i].set_low().unwrap_or(());
            }
        })
    }
}

```

Y0ba, 2022-11-28:
It would be nice, if `esp-hal` had type-erased GPIO structs like `embassy`, where you can write something like
```rust
let led1 = Output::new(p.PA0.degrade(), Level::Low, Speed::VeryHigh);
let led2 = Output::new(p.PA1.degrade(), Level::Low, Speed::VeryHigh);
```
and both variables will have the same type: `Output<'static, AnyPin>`. With type erasure it's possible to use generics in example above instead of `dyn`.

But seems like in `esp-hal` all GPIOs have their own types, so `dyn` is the only way to use them in array.

bjoernQ, 2022-11-29:
We have a larger GPIO refactoring in the works (which will take some more time) and also a smaller one which both should make it easier for us to add type-erasure .... just hold on a bit

closed 2023-02-08 by bjoernQ