esp-rs/esp-hal #279: ESP32C3: GPIO18 & 19 and JTAG USB

IsaacDynamo, 2022-11-28:
The follow code doesn't behave as expected. After the `into_push_pull_output()` call, `set_high()` and `set_low()` don't drive the pins to the correct values.

```rust
let mut leds = (
    io.pins.gpio18.into_push_pull_output(),
    io.pins.gpio19.into_push_pull_output()
);

_ = leds.0.set_high();
_ = leds.1.set_low();
```

This is caused by a default register value in the USB serial JTAG peripheral. After a reset `USB_SERIAL_JTAG_CONF0_REG.USB_SERIAL_JTAG_USB_PAD_ENABLE` is 1. This will override GPIO 18 & 19 with the USB D+ and D- signals.

In order to use GPIO 18 & 19 as GPIO this bit needs to be cleared. This logic can also be found in [esp-idf](https://github.com/espressif/esp-idf/blob/769ec42899f9de243eeb0cd31a496cb288b4110c/components/hal/esp32c3/include/hal/gpio_ll.h#L449).

For now this can be done manually with:
```rust
peripherals.USB_DEVICE.conf0.modify(|_,w| w.usb_pad_enable().clear_bit());
```

But it would be nice it this was handled by the HAL.

I can create a PR for this, that solves this in a similar way as the esp-idf does. Or would an other approach be preferred? Let me know.

An other related improvement would be to construct `UsbSerialJtag` with GPIO 18 and 19, to codify the dependency into the type system.

bjoernQ, 2022-11-29:
Good catch! Definitely something we should do in the HAL (both things mentioned)

I currently have a smaller refactoring of GPIO in the making which eliminates most of the nested macros (which should make working with the code easier) - probably worth to wait for that. Best case I can get it done today. Long term we plan a complete rewrite of the GPIO module but waiting for that probably doesn't make sense

noonien, 2023-03-20:
Got bit by this today and spent 5 hours debugging :(

bjoernQ, 2023-03-21:
Sorry to hear that. We definitely should handle this in the HAL but I'm not quite sure how we should do it API-wise: 
We cannot do it by default during start-up since that will make the HAL unusable for everyone who is exclusively using the USB (like when using the Rust-Dev-Board or S3-BOX)

MabezDev, 2023-03-21:
We can probably just do the workaround @IsaacDynamo mentioned that esp-idf uses: https://github.com/espressif/esp-idf/blob/769ec42899f9de243eeb0cd31a496cb288b4110c/components/hal/esp32c3/include/hal/gpio_ll.h#L449 and perhaps a log warning message (before actually doing it).

bjoernQ, 2023-03-21:
Oh yes - I was also thinking about using the type system (which would be possible but quite a bit of work) but the esp-idf way is much easier

PR #461 (2023-03-30): ESP32-C3: Disable `usb_pad_enable` when setting GPIO18/19 to input/output
Fixes #279
closed 2023-03-30 by jessebraham