esp-rs/esp-hal #283: [I2S] Add an option to run `i2s` with passing only `bclk` and `dout`

playfulFence, 2022-11-30:
I was testing the `esp-hal`'s `i2s` module and I had a [ESP32-C3-Lyra V2.0](https://espressif-docs.readthedocs-hosted.com/projects/esp-adf/en/latest/design-guide/dev-boards/user-guide-esp32-c3-lyra-v2.0.html) board with _**JST PH 2.0 2-Pin plug**_ (check attached photo) and noticed, that actual initialising function asks user for `ws` pin also, which is not used in this type of connection.
Actual initialising of `i2s_tx` (after creating of an `i2s` object) looks like this:
```
    let i2s_tx = i2s.i2s_tx.with_pins(PinsBclkWsDout {
        bclk: io.pins.gpio1,
        ws: io.pins.gpio20,
        dout: io.pins.gpio3,
    });
```
, where:
* bclk  => `NS_PA_CTRL` pin from [board's schematic](https://dl.espressif.com/dl/schematics/ESP32-C3-Lyra_V2_20211231.pdf)
* dout  => `PDM_TX_DATA` pin from [board's schematic](https://dl.espressif.com/dl/schematics/ESP32-C3-Lyra_V2_20211231.pdf)

`ws` may be absolutely every existing pin, since it's unused

So, maybe we should let user select his connection type and then suggest an appropriate initialisation option

[image]

tschundler, 2023-01-12:
Is that really [I2S](https://en.wikipedia.org/wiki/I%C2%B2S)? I don't think that should work correctly.

I2S needs to toggle WS to indicate a new word, which in I2S may be a variable number of bits. You can't know that every 16 bits is a new value starting from MSB again. So Clock & data alone are not enough.

That is labeled PDM, eg https://en.wikipedia.org/wiki/Pulse-density_modulation - which you could probably get close enough from high frequency PWM output. (Or I've done proper PDM before on a different uC using a frequent timer interrupt (like ~1MHz) - add the desired output value to an accumulator of the same size and set the output pin to the result of the carry - just a couple lines of assembly. Then in the main loop or another interrupt, change the target value.)

bjoernQ, 2023-01-12:
Thanks! The peripheral supports PDM but our current driver doesn't yet. We'll add support for it in future depending on the demand for those features but we are always happy to receive pull-requests adding missing features

jessebraham, 2023-12-04:
Closed by #983

closed 2023-12-04 by jessebraham