esp-rs/esp-hal #291: Questions about the ADC implementation

dimpolo, 2022-12-08:
I'd like to look into implementing ADC DMA at some point so I started by looking into the existing ADC code. There's some questions that came up:

If we make `AdcConfig.attenuations` private can this check be removed?
https://github.com/esp-rs/esp-hal/blob/f726eb05e544237fb49c44e6db058d1046f05366/esp-hal-common/src/analog/adc/xtensa.rs#L350-L355

What do these lines do? Can they be removed? What are the `sar_hall_ctrl`/ `sar_power_xpd_sar` registers? I can't find them in the reference manual.
https://github.com/esp-rs/esp-hal/blob/f726eb05e544237fb49c44e6db058d1046f05366/esp-hal-common/src/analog/adc/xtensa.rs#L282-L287
https://github.com/esp-rs/esp-hal/blob/f726eb05e544237fb49c44e6db058d1046f05366/esp-hal-common/src/analog/adc/xtensa.rs#L300-L306

Should this be wrapped in `#[cfg(esp32s2)]`? The reference manual does not mention `sar_amp_ctrl1` or `sar_amp_ctrl2` registers and the `sar_amp_ctrl3` and `sar_meas1_ctrl1` don't seem to have the fields used here. Can this be removed?
https://github.com/esp-rs/esp-hal/blob/f726eb05e544237fb49c44e6db058d1046f05366/esp-hal-common/src/analog/adc/xtensa.rs#L308-L329

bjoernQ, 2022-12-09:
In general, most times there are registers / fields not documented in the TRM the source code of `esp-idf` should contain them and usually there is also a brief description

jessebraham, 2024-01-02:
I don't think there's anything really actionable here, so going to go ahead and close this. Feel free to re-open or to open a new issue, though.

closed 2024-01-02 by jessebraham