esp-rs/esp-hal #298: Add git pre-commit hooks to check formatting

jessebraham, 2022-12-12:


igrr, 2022-12-12:
In case this helps, here are the ones I have been using: https://github.com/igrr/notred/blob/main/.pre-commit-config.yaml

bjoernQ, 2022-12-13:
Seems like rust-analyzer used something like an `xtask` for it but it also seems they don't do it anymore
The relevant PR in rust-analyzer was https://github.com/rust-lang/rust-analyzer/pull/271

I think ideally, we shouldn't rely on tools written in Python (for bare-metal) - on the other hand it's just an offer to users so it might be okay

jessebraham, 2022-12-13:
My mistake, I didn't even realize that Python tool existed. I thought the original comment in the PR was referring to the standard shell script git hooks.

PR #347 (2023-01-16): Add a simple `pre-commit` git hook to verify each package's formatting
I've left this quite simple, and it is obviously an opt-in feature. @MabezDev @bjoernQ any concerns?

Closes #298.
closed 2023-01-16 by jessebraham