esp-rs/esp-hal #300: Update linker scripts to avoid overwriting ROM function data

MabezDev, 2022-12-13:
The ESP32 specifically is a big problem here as it has its ROM function data in near the middle of the memory address space, meaning its quite easy to overwrite ROM function data on there.

bjoernQ, 2022-12-13:
Maybe we already honor it on some chips - not sure. Definitely not on ESP32

Do you have any reference how esp-idf handles it?

MabezDev, 2022-12-13:
Actually on a second look, maybe we do? I think [this section](https://github.com/esp-rs/esp-hal/blob/cfe83827cfea05fc9fa12037e5a3d63387ae9df5/esp32-hal/ld/memory.x#L20) stops us allocating `.bss` or `.data` in that ROM data region.

I think the issue me & @georgik were seeing was because of allocating a larger buffer on the stack, and the stack was overwriting the ROM function data. The question is how do we stop this? It's quite hard imo because of the dynamic nature of the size of the stack at any given point in the program.

bjoernQ, 2022-12-13:
Oh yes - I also thought about such problems before. On some chips we have _Debug Assist_ which apparently can monitor SP but not on ESP32

Also, the _Xtensa Debug Option_ doesn't seem to offer something for that

igrr, 2022-12-13:
On Xtensa chips, we can use one debug watchpoint to set a guard region at the end of the stack, covering e.g. 64 bytes. It's not perfect, but it has some probability of catching stack overflow.

On RISC-V chips, we can use the stack pointer monitor feature of debug assist, as @bjoernQ has suggested. (No IDF support to refer to, yet, but I can give some tips as I have tested this feature on FPGAs.)

closed 2024-02-19 by MabezDev