esp-rs/esp-hal #311: Consider moving our own RISCV RT overrides to its own crate

bjoernQ, 2022-12-14:
Currently the HALs targeting RISCV (ESP32-C2 and ESP32-C3 currently) include some replacements for the assembly found in `riscv-rt`

That's redundant code in both HALs - in future we will see more RISCV based chips so it makes sense to move that common functionality into its own crate

While on it we should consider if we want to remove the dependency to `riscv-rt` completely since there is very little, we use from that

Note: While initially we did that because the mentioned targets don't support `mie` and ``mip` CSRs we now also need special versions of the trap handler - that means we will continue to need it

bjoernQ, 2023-01-23:
Also, we could re-export `riscv` from there and probably can solve situations like the recent yanking of a bunch of `riscv` versions by releasing a minor update to our crate

jessebraham, 2023-01-23:
I started working on this awhile back actually, I'll dust off the project and try to get it in a repo this week some time.

jessebraham, 2023-01-25:
The package has yet to be published, but a repository has been created:  
https://github.com/esp-rs/esp-riscv-rt

Once we've tested it a bit more I will get a release out and update the necessary dependencies to use this package instead.

jessebraham, 2023-01-27:
Closed by #365

closed 2023-01-27 by jessebraham