esp-rs/esp-hal #329: Support for different crystal clock frequencies available?

rdewaele, 2023-01-03:
I would like to share that I am taking my first steps in Rust and ESP development. If this report is out of place, I apologize and will gladly accept corrections and/or (re)directions. :-)

### Issue
The HAL seems to have a hard-coded check for the crystal clock to be at 40MHz: https://github.com/esp-rs/esp-hal/blob/main/esp-hal-common/src/rtc/esp32.rs#L12
Given that this results in a panic for my ESP32 board, I suspect that this is not the only possible correct value for this clock.

### Details
I am using a [SparkFun ESP32 Thing](https://www.sparkfun.com/products/13907) and am able to flash a no-std hello world example.
However, I get a panic during startup (log has lightly edited paths).

```
Chip type:         ESP32 (revision 1)
Crystal frequency: 26MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC address:       fc:f5:c4:00:11:20
App/part. size:    100976/4128768 bytes, 2.45%
[00:00:01] ########################################      16/16      segment 0x1000                                                                                                                                                                                                                                                                                                                                                           [00:00:00] ########################################       1/1       segment 0x8000                                                                                                                                                                                                                                                                                                                                                           [00:00:03] ########################################      31/31      segment 0x10000
Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
0x40078000 - _stack_end_cpu0
    at ??:??
load:0x4007c428,len:4840
0x4007c428 - _stack_end_cpu0
    at ??:??
entry 0x4007c6a0
0x4007c6a0 - _stack_end_cpu0
    at ??:??

!! A panic occured in '~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.4.0/src/rtc/esp32.rs', at line 10, column 5

PanicInfo {
    payload: Any { .. },
    message: Some(
        assertion failed: matches!(RtcClock :: get_xtal_freq(), XtalClock :: RtcXtalFreq40M),
    ),
    location: Location {
        file: "~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.4.0/src/rtc/esp32.rs",
        line: 10,
        col: 5,
    },
    can_unwind: true,
}

Backtrace:

0x400d73de
0x400d73de - core::panicking::panic
    at ~/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:115
0x400d4453
0x400d4453 - esp_hal_common::rtc_cntl::rtc::configure_clock
    at ~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.4.0/src/rtc/esp32.rs:10
0x400d3c61
0x400d3c61 - esp_hal_common::rtc_cntl::Rtc::new
    at ~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.4.0/src/rtc_cntl.rs:96
0x400d27a2
0x400d27a2 - esp_nostd_hello_world::__xtensa_lx_rt_main
    at ~/Projects/Rust/esp-rust/esp-nostd-hello-world/src/main.rs:35
0x400d4528
0x400d4528 - Reset
    at ~/.cargo/registry/src/github.com-1ecc6299db9ec823/xtensa-lx-rt-0.14.0/src/lib.rs:77
0x400d38df
0x400d38df - ESP32Reset
    at ~/.cargo/registry/src/github.com-1ecc6299db9ec823/esp32-hal-0.7.0/src/lib.rs:86
```

line 35 in main is: `let mut rtc = Rtc::new(peripherals.RTC_CNTL);` and was generated by the esp template `cargo generate --git https://github.com/esp-rs/esp-template`
If I comment line 35 (and of course the lines that use `rtc`), the board resets every ~10 seconds, so I feel it's quite essential to have it. :sweat_smile: 

For context, some more of the main.rs code:
```
#[xtensa_lx_rt::entry]
fn main() -> ! {
    let peripherals = Peripherals::take().unwrap();
    let system = peripherals.DPORT.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    // Disable the RTC and TIMG watchdog timers
    let mut rtc = Rtc::new(peripherals.RTC_CNTL);
    let timer_group0 = TimerGroup::new(peripherals.TIMG0, &clocks);
    let mut wdt0 = timer_group0.wdt;
    let timer_group1 = TimerGroup::new(peripherals.TIMG1, &clocks);
    let mut wdt1 = timer_group1.wdt;
```

bjoernQ, 2023-01-03:
Yes, currently we only support 40MHz Xtal for ESP32 - only for ESP32-C2 we currently have a feature to select 40MHz or 26MHz

These were the changes for that: https://github.com/esp-rs/esp-hal/pull/301/files
Unfortunately, it contains some unrelated changes to the cargo runner which can be ignored

Maybe a similar approach could be used in this case - unfortunately I don't have access to any ESP32 with 26MHz Xtal

The reboots you see are because the RTC-watchdog doesn't get disabled anymore

rdewaele, 2023-01-03:
Thank you for the quick reply, and the pointers to a possible solution!

I think it would be a good learning opportunity for me to see whether I can reproduce the changes you linked to, and test them out with my board. Unfortunately, I can not commit to a time frame in which this can happen, because this work has the "when I have time" priority for now. (I am working with a wrover board, and just wondered whether the code would "just work" on this other ESP32 thing I had lying around. :-))

However, I will make sure to report back if/when I make progress in this matter.

bjoernQ, 2023-01-03:
Thank you! No rush!

bjoernQ, 2023-02-16:
Closing this since support for ESP32 26MHz Xtal was added in #394
If there is anything missing or not working as expected feel free to re-open this

closed 2023-02-16 by bjoernQ