esp-rs/esp-hal #336: embassy_hello_world panics due to cfg

Andrew-Collins, 2023-01-06:
Hi Guys, 

While working off the embassy_hello_world example for esp32_hal, I noticed that the `#[cfg(feature = "embassy-time-timg0")]` above `embassy::init` results in a panic if used in my own crate:

```
!! A panic occured in '/home/esp/.cargo/git/checkouts/esp-hal-42ec44e8c6943228/fe6ea53/esp-hal-common/src/embassy/time_driver_timg.rs', at line 32, column 37

PanicInfo {
    payload: Any { .. },
    message: Some(
        called `Option::unwrap()` on a `None` value,
    ),
    location: Location {
        file: "/home/esp/.cargo/git/checkouts/esp-hal-42ec44e8c6943228/fe6ea53/esp-hal-common/src/embassy/time_driver_timg.rs",
        line: 32,
        col: 37,
    },
    can_unwind: true,
}
 
Backtrace:
 
0x400d16fa
0x400d16fa - _ZN14esp_hal_common7embassy11time_driver12EmbassyTimer3now17hac0ec467f3d51f93E
    at ??:??
0x400d19ad
0x400d19ad - _embassy_time_now
    at ??:??
0x400d27fd
0x400d27fd - _ZN12embassy_time7instant7Instant3now17h9b9e7564e3d4cb6aE
    at ??:??
0x400d25ad
0x400d25ad - _ZN16embassy_executor3raw8Executor4poll17h77438571f7a8c994E
    at ??:??
0x400d07ac
0x400d07ac - _ZN16embassy_executor4arch8Executor3run17h14b2e8c11c7c4edaE
    at ??:??
0x400d0972
0x400d0972 - main
    at ??:??
0x400d23ba
0x400d23ba - Reset
    at ??:??
0x400d12ec
0x400d12ec - ESP32Reset
    at ??:??
```
The esp32-hal example runs without it, so can it be removed to save confusion?

Additionally while running the example I noticed: 
- The embassy example only compiles when using the `embassy-time-timg0` feature, but it is not a required-feature
- The esp32*-hal crate runners are set as `espflash flash --monitor`; I had to change it to `espflash --monitor` (which lines up with esp-template)

Happy to submit a PR if that would help.

MabezDev, 2023-01-06:
> The embassy example only compiles when using the embassy-time-timg0 feature, but it is not a required-feature

This is by design as there are multiple time driver sources (albeit only one on the esp32 right now) we test all of them in CI.

> The esp32*-hal crate runners are set as espflash flash --monitor; I had to change it to espflash --monitor (which lines up with esp-template)

espflash 2.0.0 is still a rc, but once released we'd appreciate a PR to esp-template! :)

As for the initial issue, where are you calling `embassy::init` in your code? It should be very early on, before any embassy stuff has started.

Andrew-Collins, 2023-01-07:
Running the embassy_hello_world example code in it's own crate causes the panic as well. I moved `embassy::init` (+ the cfg) as early in main as possible and it still panics. 

As above removing the cfg stops the panic; looking a bit closer it looks like the cfg causes `embassy` to be sourced from `esp-hal-common` rather than the specific hal crate (`esp32-hal` in this case): 
[image]

MabezDev, 2023-01-07:
If `embassy-time-timg0` feature is not passed then `embassy::init` will not run, so what's your full build command / project cargo.toml?

Andrew-Collins, 2023-01-07:
Repo can be found here: https://github.com/Andrew-Collins/embassy_example

Cargo.toml is 
``` 
[package]
name = "emb_rs_example"
version = "0.1.0"
authors = ["esp"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
critical-section  = "1.1.1"
xtensa-lx-rt = { version = "0.14.0", features = ["esp32"], optional = true }
esp32-hal = {package = "esp32-hal", git = "https://github.com/esp-rs/esp-hal.git", features = ["embassy", "embassy-time-timg0"] }
embassy-time  = { version = "0.1.0", features = ["nightly"]}
embassy-executor  = { package = "embassy-executor", git = "https://github.com/embassy-rs/embassy/", rev = "eed34f9", features = ["nightly", "integrated-timers"] }
esp-println       = { version = "0.3.1", features = ["esp32"] }
esp-backtrace = { version = "0.4.0", features = ["esp32", "panic-handler", "print-uart"] }
static_cell       = "1.0.0"

[features]
default = ["rt"]
rt = ["xtensa-lx-rt"]
```

Build command is: `caro run --release` (running in debug also panics)

MabezDev, 2023-01-08:
> Build command is: caro run --release (running in debug also panics)

This is the issue, you need to pass `caro run --release --features embassy-time-timg0` because of this line in _your_ crate: https://github.com/Andrew-Collins/embassy_example/blob/af1436cd344b7ce37bce1cc9266a64a9b3f5d70f/src/main.rs#L49. You've correctly enabled the feature, but you're incidentally cfging away the call to embassy::init so it doesn't work.

Andrew-Collins, 2023-01-08:
Thanks for clearing that up.
 
So when running the example from the the esp32-hal crate using:
`cargo run --release --example embassy_hello_world --features=embassy,embassy-time-timg0` 

The cfg is fine since you have the feature in the run command, but if you remove it and run the same command it still runs fine. 

So my point was just can this cfg line be removed to save confusion, but I guess it makes sense since when running the example you have to explicitly state the feature in the run command.

Andrew-Collins, 2023-01-08:
Closing, hopefully this issue also helps someone who does the same thing as I did. :)

closed 2023-01-08 by Andrew-Collins