esp-rs/esp-hal #338: Docs failing for s2

MabezDev, 2023-01-06:
https://docs.rs/crate/esp32s2-hal/latest shows the missing docs :(

jessebraham, 2023-01-09:
Here's the command that's run by the build on docs.rs:

```text
docker create \
    -v /home/cratesfyi/workspace/builds/esp32s2-hal-0.4.0/target:/opt/rustwide/target:rw,Z \
    -v /home/cratesfyi/workspace/builds/esp32s2-hal-0.4.0/source:/opt/rustwide/workdir:ro,Z \
    -v /home/cratesfyi/workspace/cargo-home:/opt/rustwide/cargo-home:ro,Z \
    -v /home/cratesfyi/workspace/rustup-home:/opt/rustwide/rustup-home:ro,Z \
    -e SOURCE_DIR=/opt/rustwide/workdir \
    -e CARGO_TARGET_DIR=/opt/rustwide/target \
    -e DOCS_RS=1 \
    -e CARGO_HOME=/opt/rustwide/cargo-home \
    -e RUSTUP_HOME=/opt/rustwide/rustup-home \
    -w /opt/rustwide/workdir \
    -m 3221225472 \
    --cpus 3 \
    --user 1001:1001 \
    --network none \
    ghcr.io/rust-lang/crates-build-env/linux@sha256:bb5b0feb4f87116a4ea905b7ff5bf5b85913db05e9f0fbc1abdd3168928d9075 \
    /opt/rustwide/cargo-home/bin/cargo +nightly rustdoc \
        --lib \
        -Zrustdoc-map \
        -Z unstable-options --config build.rustdocflags=[\"-Z\", \"unstable-options\", \"--emit=invocation-specific\", \"--resource-suffix\", \"-20221211-1.68.0-nightly-bdb07a8ec\", \"--static-root-path\", \"/-/rustdoc.static/\", \"--cap-lints\", \"warn\", \"--disable-per-crate-search\", \"--extern-html-root-takes-precedence\"] \
        --offline \
        -Zunstable-options --config=doc.extern-map.registries.crates-io=\"https://docs.rs/{pkg_name}/{version}/x86_64-unknown-linux-gnu\" \
        -j3 \
        --target x86_64-unknown-linux-gnu

```

Looks like since we don't provide any package metadata in our manifest the build is passing `--target x86_64-unknown-linux-gnu`. The error we're seeing makes sense in this case, as `usize` would be 64-bits instead of the expected 32-bits.

I'm not sure why this just started failing, I suppose there must have been an update to `xtensa-atomic-emulation-trap` since the last release? Also not entirely how how best to fix this, as we can't just pass `xtensa-esp32s2-none-elf` as the target in the package metadata.

jessebraham, 2023-01-24:
Once https://github.com/esp-rs/xtensa-atomic-emulation-trap/pull/6 is merged and we update to the latest version of `xtensa-atomic-emulation-trap` here, I think this issue _should_ be resolved.

jessebraham, 2023-01-26:
Fixed with the latest release

closed 2023-01-26 by jessebraham