esp-rs/esp-hal #346: How to ”Shared SPI access”

DaiGuard, 2023-01-15:
I refer to [esp32_hal/examples/spi_eh1_device_loopback.rs](https://github.com/esp-rs/esp-hal/blob/main/esp32-hal/examples/spi_eh1_device_loopback.rs) and try to access multiple devices on the same SPI bus.

However, I cannot access the `transfer` function from the allocated structure

- Environments
  rustc 1.66.0-nightly

- Error message

```bash
error[E0599]: the method `transfer` exists for struct `SpiBusDevice<'_, esp32_hal::esp32::SPI2, GpioPin<Output<PushPull>, Bank0GpioRegisterAccess, InputOutputPinType, 4>>`, but its trait bounds were not satisfied
   --> src/main.rs:65:22
    |
65  |         spi_device_1.transfer(&mut read, &write).unwrap();
    |                      ^^^^^^^^ method cannot be called on `SpiBusDevice<'_, esp32_hal::esp32::SPI2, GpioPin<Output<PushPull>, Bank0GpioRegisterAccess, InputOutputPinType, 4>>` due to unsatisfied trait bounds
    |
   ::: /home/dai_guard/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.4.0/src/spi.rs:861:5
    |
861 |     pub struct SpiBusDevice<'a, I, CS>
    |     ----------------------------------
    |     |
    |     doesn't satisfy `_: esp32_hal::embedded_hal::blocking::spi::transfer::Default<_>`
    |     doesn't satisfy `_: esp32_hal::prelude::_embedded_hal_blocking_spi_Transfer<_>`
    |
    = note: the following trait bounds were not satisfied:
            `SpiBusDevice<'_, esp32_hal::esp32::SPI2, GpioPin<Output<PushPull>, Bank0GpioRegisterAccess, InputOutputPinType, 4>>: esp32_hal::embedded_hal::blocking::spi::transfer::Default<_>`
            which is required by `SpiBusDevice<'_, esp32_hal::esp32::SPI2, GpioPin<Output<PushPull>, Bank0GpioRegisterAccess, InputOutputPinType, 4>>: esp32_hal::prelude::_embedded_hal_blocking_spi_Transfer<_>`
    = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
    |
4   | use embedded_hal::spi::SpiDevice;
    |

For more information about this error, try `rustc --explain E0599`.
warning: `spi_test` (bin "spi_test") generated 1 warning
error: could not compile `spi_test` due to previous error; 1 warning emitted
```

- My source code

```rust
#![no_std]
#![no_main]

use esp32_hal::{
    clock::ClockControl,
    gpio::IO,
    pac::Peripherals,
    prelude::*,
    spi::{Spi, SpiBusController, SpiBusDevice, SpiMode},
    timer::TimerGroup,
    Delay,
    Rtc,
};
use esp_backtrace as _;
use esp_println::{print, println};
// use xtensa_lx_rt::entry;

#[xtensa_lx_rt::entry]
fn main() -> ! {
    let peripherals = Peripherals::take().unwrap();
    let mut system = peripherals.DPORT.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    // Disable the RTC and TIMG watchdog timers
    let mut rtc = Rtc::new(peripherals.RTC_CNTL);
    let timer_group0 = TimerGroup::new(peripherals.TIMG0, &clocks);
    let mut wdt0 = timer_group0.wdt;
    let timer_group1 = TimerGroup::new(peripherals.TIMG1, &clocks);
    let mut wdt1 = timer_group1.wdt;

    rtc.rwdt.disable();
    wdt0.disable();
    wdt1.disable();

    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);

    let sclk = io.pins.gpio18;
    let miso = io.pins.gpio19;
    let mosi = io.pins.gpio23;

    let spi_controller = SpiBusController::from_spi(Spi::new_no_cs(
        peripherals.SPI2,
        sclk,
        mosi,
        miso,
        1000u32.kHz(),
        SpiMode::Mode3,
        &mut system.peripheral_clock_control,
        &clocks,
    ));

    let mut spi_device_1 = spi_controller.add_device(io.pins.gpio4);
    let mut spi_device_2 = spi_controller.add_device(io.pins.gpio5);

    let mut delay = Delay::new(&clocks);

    loop {

        let write= [0x2d];
        let mut read = [0x00u8];
        spi_device_1.transfer(&mut read, &write).unwrap();

        delay.delay_ms(1000u32);
    }

}
```

Please advise me what to do. Best regards.

bjoernQ, 2023-01-16:
Seems like you need to bring `SpiDevice` into scope: `use embedded_hal_1::spi::SpiDevice;`

jessebraham, 2023-01-18:
I don't think there's anything actionable here so I'm going to close this. Feel free to re-open if you disagree.

closed 2023-01-18 by jessebraham